TheRemote/Ubuntu-Server-raspi4-unofficial #3: Error on apt-get upgrade

ChrisAllisonMalta, 2019-09-02:
I'm very unfamiliar with Linux as a OS so this may be expected behaviour.

Im using the v2 of the image released on Aug 31.

After I run sudo apt-get update && sudo apt-get upgrade after lots of lines get printed I get a bunch of messages shown in the attached photo

[image]

I was getting a similar error when I ran v2 of the disk image so I guess its expected behaviour as you don't want the ubuntu firmware to update on this device?

TheRemote, 2019-09-02:
Hey Chris, 

That's exactly right.  The firmware in the Ubuntu repository is unfortunately the firmware that does not contain RPI4 support (basically, flash-kernel replaces /boot/firmware with the version so it won't boot afterward).  Once Ubuntu updates it will be safe for flash-kernel to run again

closed 2019-09-02 by TheRemote

ChrisAllisonMalta, 2019-09-02:
Thanks @TheRemote , enjoy Labour day