TheRemote/Ubuntu-Server-raspi4-unofficial #4: Unable to reproduce image build from script

damarshall, 2019-09-02:
Thanks for putting this together.
I've been working on reproducing the image from the Build script, however it doesn't work.
I was able to reverse engineer some aspects not covered in the script, such as copying start4*.elf into /boot/firmware and placing the cmdline.txt and config.txt in bootfiles so it can be copied as part of the build.

However, my kernel builds result in a kernel panic.
Are there other steps you are taking not documented in the Build script?

thanks

damarshall, 2019-09-02:
...also, there's no difference in the Build script between v2 and v3.

I just tested v3 it works well with a single microSD card. However if I modify cmdline.txt and use a usbroot   (root=/dev/sda2) there's a kernel panic.

I'll help work on this but need to be able to reproduce the build.
Cheers

TheRemote, 2019-09-03:
Hey damarshall,

The script is missing an explanation for a few things (such as where to get the bootfiles folder from, a couple extra files, etc.).  I'm going to leave this issue open until I can write this up.

damarshall, 2019-09-03:
thanks - much more in latest script.
Can you check in ./conform_config.sh as I think that's key

TheRemote, 2019-09-03:
Ahh absolutely, so conform_config is part of sakaki-'s precompiled set of 64 bit kernel modifications.  Specifically here at this repo:

https://github.com/sakaki-/bcmrpi3-kernel-bis/blob/master/conform_config.sh

This script configures a bunch of flags in the kernel .config enabling things we want for Ubuntu to run properly.

For bootfiles you can basically just copy the files from the /boot partition of any of the images.  Some of them are updated from compiled files like the dtbos but for the most part they are static.

I got the two packages to install haveged by just installing them on a Pi and then grabbing them from the apt cache.

I think that's about it, I'm at work right now so if I missed something let me know and I will explain it.  I also recently published my main post about this which is https://jamesachambers.com/raspberry-pi-4-ubuntu-server-desktop-18-04-3-image-unofficial/.  This page will contain a complete build walkthrough that I'm going to write over the next day or two.

damarshall, 2019-09-08:
Am able to reproduce the build - thanks for all the improvements

closed 2019-09-08 by damarshall