TheRemote/Ubuntu-Server-raspi4-unofficial #6: Nonvisible GUI (KDE Plasma?)

mrystrand, 2019-09-04:
Just got this installed and understand this is not officially supported.
However, I am experiencing some quite major issues with the GUI.
In the attached screenshot you can see the start menu for Kubuntu.
Running on a RPi4 4GB with nothing else connected than USB keyboard, mouse and an HDMI display.

Anything I can do to improve or resolve this?

[image]

TheRemote, 2019-09-05:
Hmmm, we need to get some log files for this.  It's probably not going to be that hard to fix once we know what is going wrong.

What you could do is get the logs off the SD card by plugging it into your computer and going to /var/log/ which should contain the information we need.

TheRemote, 2019-09-07:
This has been fixed in v6

closed 2019-09-07 by TheRemote