TheRemote/Ubuntu-Server-raspi4-unofficial #9: RDB is missing (was Ceph is missing)

dkrizic, 2019-09-07:
Hi, 

thank you very much for the kernel with Ceph, but unfortunately RDB is still not supported. My rook-ceph-agent reports:

2019-09-07 19:50:07.898118 I | exec: Running command: modprobe rbd
2019-09-07 19:50:07.905521 I | modprobe rbd: modprobe: FATAL: Module rbd not found.

I thought, that RDB is part of Ceph, and I can see that Ceph is in the Kernel, but RDB is not.

dkrizic, 2019-09-07:
Some documentations are stating, that rdb should be part of the kernel as of 4.x. I am confused.

dkrizic, 2019-09-07:
From rook-ceph documentation:

Re-configure your kernel to make sure the CONFIG_BLK_DEV_RBD=y in the .config file, then build the kernel.

https://github.com/rook/rook/blob/master/Documentation/ceph-common-issues.md#rook-agent-rbd-module-missing-error

charles-d-burton, 2019-09-07:
I'm seeing the same issue.

TheRemote, 2019-09-08:
Thank you, this helps a lot since I didn't know exactly what this was but that link tells me everything I need!

dkrizic, 2019-09-08:
A little background what I am doing: I have a Kubernetes cluster on bare-bone APU2C4 boards (amd64, 4 cores, low power). Each of these boards have a 500GB SSD card. 100GB is used for the OS, 400 for Ceph. I run Kubernetes 1.15.3 with Calico networking.

I run rook-ceph on the cluster that provides redundant storage; so the cluster is self converged.

I would like to extend this cluster with a few Raspberry Pi 4 nodes. Meanwhile a lot of things are working:

- [x] Calico CNI daemonset runs (they have an arm64 image)
- [x] Rook Ceph Discovery runs
- [ ] Rook Ceph Agent does not run since it requires RBD kernel module

With the next kernel that includes RBD the Rook Ceph Agent should start and be able to mount Ceph volumes on the Raspi nodes.

dkrizic, 2019-09-08:
When is the next run/release expected?

TheRemote, 2019-09-08:
I've committed this in v7, can someone give it a try?

dkrizic, 2019-09-08:
Testing

dkrizic, 2019-09-08:
RBD is here! rook-ceph starts. Thanks!

State: I am unable to mount volumes to the Raspi node (which does not provide any storage, just consumes), but this seems to be a rook-ceph/Ceph specific problem.

closed 2019-09-08 by dkrizic

dkrizic, 2019-09-08:
One observation: The fan of the HAT does not run anymore... I should have run long ago... I don't know if this has to do with the kernel.