TheRemote/Ubuntu-Server-raspi4-unofficial #10: Serial port (ttyS0) does not appear

jcgarciaca, 2019-09-08:
I tried with image V6 and it works quite well.
However, I need to use serial communications with GPIO, but I noted that `/dev/ttyS0` does not appear. How could I disable serial console in order to use serial communication?
Thank you.

yashiki2005, 2019-09-10:
On Raspberry Pi4, uart0 is reserved for Bluetooth. Same as Raspberry Pi3 Model B.
You can use uart if you disable Bluetooth or enable miniuart.

Method:
Add the following string to /boot/firmware/config.txt:
dtoverlay = miniuart-bt
core_freq = 250

Next, you need to disable the serial console.
$ sudo systemctl stop serial-getty@ttyAMA0.service
$ sudo systemctl disable serial-getty@ttyAMA0.service

Edit /boot/firmware/cmdline.txt and delete strings such as "console = serial0,115200"

Next, stop Pi4.
Connect GPIO14 and GPIO15 and turn on Pi4.

This procedure disables the serial console and enables uart (available for serial communication) on GPIO14 (TX) and GPIO15 (RX). Bluetooth is also available.

jcgarciaca, 2019-09-10:
@yashiki2005 
Thank you for your answer.
Just for reference, I removed `ttyAMA0` references in `cmdline.txt` and now I am able to use serial communication.