TheRemote/Ubuntu-Server-raspi4-unofficial #11: Separate release asset(s) for kernel, firmware, etc.

mwpastore, 2019-09-09:
I know this installer is a temporary solution, but in the short term it would be super-helpful to have a separate release asset(s) for the kernel, firmware, etc., including stock/default versions of cmdline.txt and config.txt. 

I've been `losetup`'ing the whole .img and copying stuff out to apply updates. Not fun!

TheRemote, 2019-09-09:
Hey Mike,

I completely agree.  I may just commit the entire build tree at this point so it's easy for anyone to tweak and we have no idea how long of a wait it may be until official support comes.

TheRemote, 2019-10-05:
This is in the process of being added, config.txt and those are now in here as well as firmware files that aren't found in firmware-nonfree.  This will continue to expand and once it leaves pre-release all of the sections will be in there.

closed 2019-10-05 by TheRemote