TheRemote/Ubuntu-Server-raspi4-unofficial #12: ZFS kernel module not found

AnotherDroog, 2019-09-10:
`Sep 10 07:07:07 ubuntu modprobe[24588]: modprobe: FATAL: Module zfs not found in directory /lib/modules`

AnotherDroog, 2019-09-10:
Typically installable via `sudo apt install zfs-dkms` however, kernel headers are supposedly unavailable although linux-headers-raspi2 is installed.

```Module build for kernel 4.19.71-v8+ was skipped since the
kernel headers for this kernel does not seem to be installed```

AnotherDroog, 2019-09-10:
Even happens after running `make headers_install` from raspi-kernel 4.19.y branch

AnotherDroog, 2019-09-10:
Okay, I've solved it. Would you be interested in a PR?

yashiki2005, 2019-09-10:
On Raspberry Pi4, linux-headers-raspi2 is not correct headers. linux-headers-raspi2 is for armhf, But Ubuntu-Server-raspi4-unofficial running aarch64.
Also spl-dkms and zfs-dkms is obsolete version. Some modification is required for work(change "timespec" to "timespec64" etc...).

1.check kernel header version = running os version
$ cat /lib/modules/`uname -r`/build/include/generated/utsrelease.h
$ uname -r
If version string is different or error occurred, You must recompile kernel with current kernel config (stored at       /boot/firmware/config)
1.5 recompile kernel and boot with new kernel
$ git clone https://github.com/raspberrypi/linux.git  -b rpi-4.19.y
$ cd linux
$ cp /boot/firmware/config .config
$ make oldconfig
$ make -j4 Image modules 
$ sudo cp arch/arm64/boot/Image /boot/firmware/system8.img
$ sudo make modules_install
$ sudo reboot
and re-try procedure 1.

2. get latest zfs from github and build it
$ git clone https://github.com/zfsonlinux/zfs.git
$ cd zfs
$ ./autogen.sh
$ ./configure && make -j4 && make check && make install

after install, zfs.ko and spl.ko is located under /lib/modules/`uname -r`/extra/

AnotherDroog, 2019-09-11:
Thank you, those instructions are useful.

Have you also noticed ZFS native encryption is not working?

yashiki2005, 2019-09-11:
zfs-dkms (stable branch) may have encryption disabled by default. To encrypt, dm-crypt and zfs must be used together.
zfs at github.com (master branch ver 0.8.0) has encryption enabled.

To verify that zfs supports encryption,
$ grep -i encrypt /usr/include/libzfs/sys/fs/zfs.h
If ZFS_PROP_ENCRYPTION is displayed, zfs supports encryption.
Also,
# zfs get encryption
Indicates that encryption is enabled or not supported.
NAME PROPERTY VALUE SOURCE
test encryption off default
test / vol1 encryption aes-256-ccm-
It will be displayed like this.

I created a zpool for the file on a trial basis, created zvol with zfs create encryption = on and wrote the file, but as far as I saw with hexdump, encryption seems to work.

AnotherDroog, 2019-09-11:
Thank you that did it!

TheRemote, 2019-09-18:
These are really good instructions.  I'm going to leave this open until I release an image with working kernel headers included.  Thanks everyone!

yashiki2005, 2019-09-19:
When compiling the kernel with CONFIG_IKHEADERS = y, the kernel header tarball is placed in /sys/kernel/kheaders.tar.xz
If you compile the kernel with CONFIG_IKHEADERS = m, run sudo modprobe kheaders to get the kernel header tarball. The kernel header tarball is located at /sys/kernel/kheaders.tar.xz.

This kheaders.tar.xz is compatible with the currently running kernel. It works even if it is expanded under /usr/include.

Since CONFIG_IKHEADERS is not defined in /boot/ firmware/config(raspi4 Unofficial Image config), kernel headers compatible with the current kernel can be easily obtained by setting y or m.

closed 2019-09-26 by TheRemote