TheRemote/Ubuntu-Server-raspi4-unofficial #15: PoE HAT fan stopped working in pre-release v7

christian-schlichtherle, 2019-09-12:
First of all, thanks for this awesome work. I'm using your images for running an array of Raspberry Pi 4B with Kubernetes. I'm writing a blog entry for this, currently.

I've been following your prereleases and trying most of them successfully.
For me, the last-known good image was pre-release v6.
In pre-release v7, the fan of my PoE HAT doesn't work, which throttles the CPU eventually.

TheRemote, 2019-09-18:
That's very strange, I will do some investigating but I don't have the PoE hat so it might be a bit tricky but let me do some research and see what it turns up and what we may have changed that would have done this!

christian-schlichtherle, 2019-09-18:
BTW: I do get the same issue

DaftHonk, 2019-09-23:
Can confirm. Running pre-release v8 on Raspberry Pi 4B (4GB).
Fortunately I'm using mine as a Unifi Controller so I likely won't see too much throttling under normal use.

christian-schlichtherle, 2019-09-23:
> BTW: I do get the same issue when I use the official Raspbian Lite and then do `rpi-update` and then edit `/boot/config.txt` to contain `arm_64bit=1`. Maybe the issue is specific to the kernel changes from Raspbian?

The problem has been fixed with the latest `rpi-update`. However, I'm not using Raspbian for now, so I really hope for a solution to this issue.

christian-schlichtherle, 2019-09-24:
Here are some more insights: I'm running a Kubernetes cluster on an RPi 4 array of seven nodes:

```
$ kubectl get nodes -o wide
NAME     STATUS   ROLES    AGE   VERSION         INTERNAL-IP     EXTERNAL-IP   OS-IMAGE                         KERNEL-VERSION   CONTAINER-RUNTIME
node-0   Ready    worker   43m   v1.15.4-k3s.1   192.168.71.20   <none>        Ubuntu 18.04.3 LTS               4.19.73-v8+      docker://19.3.2
node-1   Ready    master   44m   v1.15.4-k3s.1   192.168.71.21   <none>        Ubuntu 18.04.3 LTS               4.19.71-v8+      docker://19.3.2
node-2   Ready    worker   44m   v1.15.4-k3s.1   192.168.71.22   <none>        Ubuntu 18.04.3 LTS               4.19.71-v8+      docker://19.3.2
node-3   Ready    worker   43m   v1.15.4-k3s.1   192.168.71.23   <none>        Ubuntu 18.04.3 LTS               4.19.71-v8+      docker://19.3.2
node-4   Ready    worker   44m   v1.15.4-k3s.1   192.168.71.24   <none>        Ubuntu 18.04.3 LTS               4.19.71-v8+      docker://19.3.2
node-5   Ready    worker   44m   v1.15.4-k3s.1   192.168.71.25   <none>        Ubuntu 18.04.3 LTS               4.19.71-v8+      docker://19.3.2
node-6   Ready    worker   43m   v1.15.4-k3s.1   192.168.71.26   <none>        Raspbian GNU/Linux 10 (buster)   4.19.73-v8+      docker://19.3.2
```

`node-0` is running the v8 image from this repo. `node-6` is running the latest Raspbian Lite with the kernel patched via `rpi-update`, as decribed [here](https://illegalexception.schlichtherle.de/raspbian/2019/09/21/how-to-upgrade-raspbian-to-a-64-bit-kernel/). The other five nodes are running the v6 image from this repo.

As you can see, `node-0` and `node-6` have the same kernel version, which is 4.19.73-v8+.
However, the fan of the PoE HAT on `node-0` is not working, so when observing the temperatures, I get this:

```
$ ansible all -m shell -a 'vcgencmd measure_temp; vcgencmd measure_clock arm; vcgencmd get_throttled' --one-line | sort
node-0 | CHANGED | rc=0 | (stdout) temp=85.0'C\nfrequency(48)=500132800\nthrottled=0x60006
node-1 | CHANGED | rc=0 | (stdout) temp=50.0'C\nfrequency(48)=1750412160\nthrottled=0x0
node-2 | CHANGED | rc=0 | (stdout) temp=79.0'C\nfrequency(48)=1750412160\nthrottled=0x0
node-3 | CHANGED | rc=0 | (stdout) temp=69.0'C\nfrequency(48)=1750464896\nthrottled=0x0
node-4 | CHANGED | rc=0 | (stdout) temp=71.0'C\nfrequency(48)=1750464896\nthrottled=0x0
node-5 | CHANGED | rc=0 | (stdout) temp=73.0'C\nfrequency(48)=1750412160\nthrottled=0x0
node-6 | CHANGED | rc=0 | (stdout) temp=73.0'C\nfrequency(48)=1750464896\nthrottled=0x0
```

Clearly, `node-0` is unhealthy. I hope this information will help you track down the root cause.

damarshall, 2019-09-24:
in the current __.config__ the correct module is not set, we have:
`# CONFIG_SENSORS_RPI_POE_FAN is not set`

It is set upstream on Sakaki-'s __conform_config.sh__ where the following is found:

`# Enable RPI POE HAT fan`
`set_kernel_config CONFIG_SENSORS_RPI_POE_FAN m`

This should pick up automatically next build

TheRemote, 2019-10-03:
This should be fixed in v9 -- can you verify?

christian-schlichtherle, 2019-10-03:
PoE HAT is working - superb!
I think this issue can be closed now.

closed 2019-10-03 by christian-schlichtherle