TheRemote/Ubuntu-Server-raspi4-unofficial #16: 5GHz doesn't work on wifi

itsermo, 2019-09-14:
I'm unable to connect to my 5GHz network, despite that Raspberry Pi 4 has support for this.  Is there a setting in the boot args or something that turns on 5GHz support for the Broadcom wifi chip?

yashiki2005, 2019-09-15:
I use 802.11ac (5GHz) without problem. There are no additional settings or startup arguments. 
Connect wifi using netplan.

Try connecting to WiFi manually:
$ ps -ax | grep wpa 
If there are other wpa_supplicant processes, stop them.
$ sudo wpa_passphrase SSID WPAPASSWORD | sudo tee /etc/wpa_supplicant/wpa_supplicant.conf
$ sudo wpa_supplicant -iwlan0 -dd -c/etc/wpa_supplicant/ wpa_supplicant.conf
wpa_supplicant displays the wifi negotiation status/debug info. If you see an error, show it and you might be able to fix it.

Also, raspberry pi4b/3b is in a disabled state until it is set in which country it is used. It is better to check if you can connect to 5GHz wifi with raspbian first.

itsermo, 2019-09-16:
never mind, it seems to work now, not sure why it wasn't connecting before or seeing my 5GHz network, although I was able to connect to my other 2.4GHz SSID, no problem. perhaps an unknown bug in the driver?

itsermo, 2020-05-01:
I discovered that the issue I was having was due to not having set the regulatory domain for wifi in the crda defaults. my 5ghz network channel appeared to be on the boundaries of what is allowed when crda is unset

you can get this information by typing:
`iw reg get`

to change it you can use the command, if you are in the US, for example:
`iw reg set US`

however this change will be lost after reboot, so the best way is to make sure you have crda installed

`sudo apt-get install crda`

and edit the crda default file

`sudo vi /etc/crda`
`REGDOMAIN=US`

these changes will stick after reboot.

is there any way that this can be set to default (US) or informing people on first boot that this field must be set?

closed 2022-05-11 by TheRemote