TheRemote/Ubuntu-Server-raspi4-unofficial #17: Problem with install net core

KrzysztofPajak, 2019-09-15:
After fresh instalation I try install net core based on documentation:
https://dotnet.microsoft.com/download/linux-package-manager/ubuntu18-04/sdk-current
Unfortunetly is not possible, I'got message:
"E: Couldn't find any package by glob 'dotnet-sdk-2.2y'
E: Couldn't find any package by regex 'dotnet-sdk-2.2y'"

Where could be the problem, thanks for ony help?
Regards
Krzysztof

yashiki2005, 2019-09-15:
dotnet-sdk-2.2.x for arm64 is not supplied via apt. apt-get install dotnet-sdk-2.2 is for x86_64 only(not for arm or arm64).
dotnet SDK 2.2.402 for arm64 is here: https://dotnet.microsoft.com/download/thank-you/dotnet-sdk-2.2.402-linux-arm64-binaries. You will download dotnet-sdk-2.2.402-linux-arm64.tar.gz
then,
$ tar xvf dotnet-sdk-2.2.402-linux-arm64.tar.gz
and launch dotnet
$ ./dotnet
you see dotnet help message.

Note:
donet on github: https://github.com/dotnet/core
dotnet 3.0 core is here(under developmevnt): https://gist.github.com/richlander/467813274cea8abc624553ee72b28213

TheRemote, 2019-09-18:
Thanks yashiki2005!

closed 2019-09-18 by TheRemote