TheRemote/Ubuntu-Server-raspi4-unofficial #18: strange behavior with arm_freq=2000

wtarreau, 2019-09-17:
Hi,

I used a previous version (v2 I guess) in which I had set arm_freq=2000 in config.txt, which worked pretty well. With v7, it still boots and reports 2.0 GHz in cpufreq but really runs at 1.0 GHz only, despite this frequency not being available in cpufreq. Latest raspbian image with the v8 kernel 4.9.71 works fine. I picked this kernel and replaced the current one with it, and it had the same behavior. The DTB being the same, I suspect the issue is within the boot loader itself, but at this point I failed to try to guess what other file affects this because whetever else I change results in a non-booting system. Thus I'm wondering if there could be some hard-coded dvfs op points in the boot loader for example.

It's not critical, I tried on a different flash so I can stay on the previous one for now. Thanks!

tommythorn, 2019-09-18:
A related issue I believe: while I can install `perf`, it doesn't work because it needs the exact kernel version support (sorry, I don't have access to the RPi 4 atm so relying on fuzzy memory).  If this is easy to fix that would be great to have.

wtarreau, 2019-09-18:
I finally managed to get it to boot with the kernel8.img from raspbian. I iteratively removed a number of apparently unused files from /boot/firmware, and my system still works fine with only this there:
```
COPYING.linux
LICENCE.broadcom
bcm2711-rpi-4-b.dtb
bootcode.bin
cmdline.txt
config.txt
fixup4.dat
fixup4cd.dat
fixup4db.dat
fixup4x.dat
issue.txt
kernel8.img
overlays
start4.elf
start4cd.elf
start4db.elf
start4x.elf
```
I compared some files, and the dtb is the same, thus I'm assuming that only the kernel makes the difference. Why it didn't boot when picked alone is still unknown to me (maybe the initrd, maybe the presence of armstub8-gic which is not on the raspbian image conflicts with their kernel). However now that confirms that what makes the SoC work at 1 GHz instead of 2 GHz definitely seems to be the kernel itself since changing it fixes this.

I had to set `root=/dev/mmcblk0p2` in cmdline.txt of course, since the raspbian's version references a UUID. My `config.txt` file now has this:
```
[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
# dtoverlay=vc4-fkms-v3d
max_framebuffers=2
over_voltage=6
arm_freq=2000
gpu_mem=16

[all]
enable_gic=1
arm_64bit=1
```

Output of the mhz utility:
```
ubuntu@ubuntu:~$ ./mhz 100
us=55723 cpu_MHz=1794
us=50121 cpu_MHz=1995
us=50041 cpu_MHz=1998
us=50023 cpu_MHz=1999
us=50029 cpu_MHz=1998
us=50034 cpu_MHz=1998
us=50044 cpu_MHz=1998
us=50021 cpu_MHz=1999
```

wtarreau, 2019-09-18:
Oh and I found there that I could also remove `bootcode.bin`, and the system indeed continues to work : https://www.raspberrypi.org/documentation/hardware/raspberrypi/booteeprom.md

I think it's useful to remove confusing unused parts, it's less difficulty to troubleshoot a non-booting system.

TheRemote, 2019-09-18:
Hey wtarreau,

These are really good points.  The difference was indeed the armstub8-gic.bin as that file is one that we build custom for the Raspberry Pi 4.

bootcode.bin is no longer used on the Pi 4 at all as the Pi 4 is the first one that has a on-board flashable bootloader built into the device.

I agree some of the legacy files should be removed as this image is not intended to work on anything but a Pi 4.

closed 2019-09-18 by TheRemote