TheRemote/Ubuntu-Server-raspi4-unofficial #19: `perf` doesn't work

tommythorn, 2019-09-18:
`sudo apt install perf` succeeds, but the resulting `perf` command doesn't work:
```
ubuntu@ubuntu:~$ perf ls
WARNING: perf not found for kernel 4.19.69

  You may need to install the following packages for this specific kernel:
    linux-tools-4.19.69-v8+
    linux-cloud-tools-4.19.69-v8+

  You may also want to install one of the following packages to keep up to date:
    linux-tools-v8+
    linux-cloud-tools-v8+
ubuntu@ubuntu:~$ sudo apt install     linux-tools-4.19.69-v8+
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package linux-tools-4.19.69-v8
E: Couldn't find any package by glob 'linux-tools-4.19.69-v8'
E: Couldn't find any package by regex 'linux-tools-4.19.69-v8'
```

I hope we can fix this

wtarreau, 2019-09-18:
I agree that it's convenient, especially on A72 which does have a number of hardware events. For a reason I don't know, perf is always a mess on ubuntu, it seems to tightly related to the installed kernel version. I managed to install it by hand by picking the closest one, and it works fine for me:
```
# apt install linux-tools-4.18.0-25-generic
# /usr/lib/linux-tools/4.18.0-25-generic/perf top
```
Hoping this helps.

tommythorn, 2019-09-18:
Brilliant, thanks a bunch!

closed 2019-09-18 by tommythorn