TheRemote/Ubuntu-Server-raspi4-unofficial #32: any way to upgrade release versions?

itsermo, 2019-09-28:
if I want to take advantage of the new releases, is my only option to format my SD card and start from scratch?

my system is stable now, and I'd hate to start from scratch.  I want the benefits of new kernel and whatever changes are made between these unofficial releases.

TheRemote, 2019-09-28:
This is coming very soon as we exit the pre-release stage.

My plan is to put the kernel/modules binaries inside this repository and have a modified version of the build script that simply pulls the kernel and updates the modules as well as the Raspberry Pi firmware.  This would cover 95% of the changes made over all the releases (the rest of them were small configuration file changes early on).

I think there will be one or two more pre-releases as I finish closing out the open issues in the GitHub and then this will be a part of all releases after pre-release ends!

Thanks for the feedback.  I know others are looking for this as well.

TheRemote, 2019-10-28:
This is finally completed in v15.  

Make sure you do a backup of anything important first.  Here is how you do it:

```
wget https://raw.githubusercontent.com/TheRemote/Ubuntu-Server-raspi4-unofficial/master/Updater.sh
chmod +x Updater.sh
sudo ./Updater.sh
```

Make sure you sudo when you use it, but it's that simple.  It will check if there is a new version first before trying anything but if you just want to get your images up to date you can run it then delete it.

In new versions of the image it's already built in to the /home folder.:

`sudo bash /home/Updater.sh`

closed 2019-10-28 by TheRemote