TheRemote/Ubuntu-Server-raspi4-unofficial #33: Unit shuts down from video playback

johnjelinek, 2019-09-30:
In v8 pre-release, if I play a YouTube video at fullscreen, it powers off. This happens on other video players as well. I'm using the 4GB RAM model.

TheRemote, 2019-10-03:
Does this happen immediately or does it play for a little bit first?

The GPU firmware was updated for v9 as well as a bunch of hardware kernel config changes so it may be worth giving it another go.

I'll be testing some more in the desktop version tonight as well and I'll make sure to test this.

johnjelinek, 2019-10-03:
It plays a little bit first, but it seems to run hotter overall in Ubuntu
than Raspbian. I learned that with the official case, I can get it to
overheat and shut down in Raspbian as well. I switched to a Flirc case and
now it works fine in Raspbian with video playback. I can give v9 a shot
with the Flirc case.

TheRemote, 2019-10-04:
If you have time try with v10

TheRemote, 2019-10-04:
I was able to reproduce this as well.  I believe the solution to this is to get the v3d driver working properly in the image.

I posted a lot more about where I'm at in that process here: https://github.com/TheRemote/Ubuntu-Server-raspi4-unofficial/issues/36

TheRemote, 2019-10-05:
This has been fixed in v11

closed 2019-10-05 by TheRemote