TheRemote/Ubuntu-Server-raspi4-unofficial #36: Performance

antonical, 2019-10-01:
Hi I am trying your latest V8 image. I have also installed the full desktop to mirror what we a have on a Rpi3 B+. It runs fine and all application seems to run ok. But....

The RPi4 seems slow graphically compared to the Rpi3 B+. The #dtoverlay=vc4-fkms-v3d is set in config.txt file. Which is a surprise given the graphics chip, memory (4GB model) and cpu increases of the Rpi4.

Any advice or suggestions welcome.

Cheers
Spart

small-mallet, 2019-10-01:
Graphics in 64 bit kernel is a long way away from being useful/good. Try this first: https://github.com/sakaki-/raspbian-nspawn-64

If the graphical performance is as good as the offical raspbian 32 bit (while running from both 32-bit userland AND 64-bit nspawn container), it might have something to do with Ubuntu not being optimized like raspbian is.

tommythorn, 2019-10-01:
I haven't actually measured it, but I immediately noticed that graphics on the RPi4, running the *official* Raspbian, seemed slower than on RPi3.  Switching back to the legacy driver (the non-GL), made it a bit better but still seemed slow.  AFAICT, Ubuntu is the same as Raspbian on the same hardware.

antonical, 2019-10-01:
@tommythorn Yes I noticed exactly that. The Rpi3 is a lot snappier and not mouse laggy.

Cheers
Spart

TheRemote, 2019-10-03:
They're still changing a ton of kernel build flags to better support the Pi 4 hardware.  They enabled a ton of new ones in the latest default configs.  I expect this will continue to keep improving in the short term pretty quickly.

Can you try v9 for comparison?

antonical, 2019-10-03:
Wilco. I will report back as soon as I am able.

TheRemote, 2019-10-04:
So here is where I am at with this.  Performance can be greatly increased if you uncomment the #dtoverlay=vc4-fkms-v3d in /boot/firmware/config.txt.

The problem I'm seeing is that there's a lot of graphical glitches (artifacts around the borders of the windows, title bar does not appear to be drawn correctly).  However, if I stumble into a terminal and run the glxgears demo I am easily hitting over 1000 fps on my Pi 4.

The problem is likely some simple configuration item but narrowing it down is taking some time.  To get max performance the image needs the v3d driver enabled.  I know sakiki- was able to do this in his 64 bit Gentoo distro so it's definitely possible, I'm just still buried going through documentation and code trying to figure out what the magic config is!

TheRemote, 2019-10-05:
This has been dramatically improved in v11.

Note for Kubuntu desktop: the compositor needs to be disabled for everything to draw correctly.  Use the shortcut Alt + Shift + F12 to disable the compositor and everything will draw normally.  Now go into the Settings app and go to the "Compositor" section and uncheck "Enable compositor at startup" to prevent it from turning back on when you reboot.

closed 2019-10-05 by TheRemote