TheRemote/Ubuntu-Server-raspi4-unofficial #40: graphical glitches in desktop V9

MadWomen, 2019-10-03:
i get graphical glitches in kubuntu desktop using pre release v9.
some windows appear to be completely empty.

tested this on a pi 4B 2gb and 4gb.

i'm going to test this on xubuntu aswell in a moment.

TheRemote, 2019-10-04:
Thanks, my full desktop install completed overnight, I'll begin investigating

TheRemote, 2019-10-04:
I found some problems, can you test again with v10?

MadWomen, 2019-10-04:
will do!

MadWomen, 2019-10-04:
the graphical glitches are still there unfortunatly in v10, the wifi works now though!

TheRemote, 2019-10-04:
Thanks for testing it!

I posted a more lengthy explanation on where I'm at on the v3d driver in this thread: https://github.com/TheRemote/Ubuntu-Server-raspi4-unofficial/issues/36

TheRemote, 2019-10-05:
This has been fixed in v11.

Note for Kubuntu desktop: the compositor needs to be disabled for everything to draw correctly.  Use the shortcut Alt + Shift + F12 to disable the compositor and everything will draw normally.  Now go into the Settings app and go to the "Compositor" section and uncheck "Enable compositor at startup" to prevent it from turning back on when you reboot.

I'm working on a way to roll the Kubuntu fix in the image so the compositor is off by default but it's an easy fix now that the 3d driver is working!

closed 2019-10-05 by TheRemote