TheRemote/Ubuntu-Server-raspi4-unofficial #42: 16.04?

peardox, 2019-10-04:
Would a 16.04 version be relatively easy to make?

I'm doing build tests that mostly compile on 18.04 but the master is 16.04 so having access to both would be rather useful

TheRemote, 2019-10-05:
I think it would probably work.  It may need a tweak or two but presumably you could just put the 16.04 image .xz in the place of the 18.04 image.  The only difference should be the versions of the Ubuntu packages themselves.

You can also try it as well if you have some bash understanding but I will definitely give you an update when I try it!

peardox, 2019-10-05:
Yeah - I'll give it a go with 16.04

I'm running tests on JavaCPP and 18.04 worked perfectly on your build - I already know the next stage of this will fail (JavaCPP-Preset problem, nothing to do with yours)

I believe that the bit that should fail will work on 16.04 (libc removed some stuff)

peardox, 2019-10-05:
Oh - just read the (rest of the) install instructions

Running on a SSD / HD...

Root pivot is a good way to use the SD only for boot - explanation starts here - https://www.jetsonhacks.com/2019/09/17/jetson-nano-run-from-usb-drive/

A Jetson Nano is essentially a Pi3B+ with 4G and a GPU - I run mine with the method described and it works really well.

TheRemote, 2019-10-24:
So it took me a while but I finally tried building this with 16.04 but there's a problem.  It's only available in armhf (32 bit) and my whole script is set up to do 64 bit.

There is an installable version but I looked inside the iso for it and it doesn't work anything like the preinstalled images unfortunately!

The Jetson Nano looks really interesting.  It's the first time I've seen it!  It sounds like it's meant for AI workloads but I'm sure it's a lot of fun to play with!

closed 2019-10-24 by TheRemote