TheRemote/Ubuntu-Server-raspi4-unofficial #43: bluetooth not working

phmatos, 2019-10-05:
Hello the v11 system is not working the bluetooth adapter, i have a pi 4gb model

darkhorseman45, 2019-10-05:
I have the same problem.. Same system raspberry pi4 4gig model. The bluetooth is not working installed kubuntu desktop everything went well. Wireless working youtube is great very snappy, but when i go to the bluetooth section it says adapter not found.

darkhorseman45, 2019-10-05:
[image]

darkhorseman45, 2019-10-05:
By the way saw all the work you did yesterday. I really appreciate you thank you for the hard work

MadWomen, 2019-10-05:
can confirm, not working in v11

TheRemote, 2019-10-06:
Hey everyone, thanks for the reports.  I swear I had this working but I was so excited about the video driver fix that apparently I did not test well enough!

I will investigate and report back!

TheRemote, 2019-10-06:
I found it, can you try with v12?

MadWomen, 2019-10-06:
bluetooth works now in v12, thank you for the quick fix!

TheRemote, 2019-10-06:
No problem!

closed 2019-10-06 by TheRemote