TheRemote/Ubuntu-Server-raspi4-unofficial #44: Sound is garbled

johnjelinek, 2019-10-05:
I'm running v11. If I boot into Ubuntu Mate desktop, open a terminal, and run `say hello`, the sound becomes crackly and garbled. All other sound is distorted until I reboot.

johnjelinek, 2019-10-05:
In other news, using a Flirc case on v11 gives me full 4k resolution and I can play choppy YouTube videos fullscreen without the Pi shutting off.

TheRemote, 2019-10-06:
Can you try this in v12?
I brought over all the Raspbian firmware and a bunch of stuff was updated that was not up to date in the firmware-nonfree repository

MadWomen, 2019-10-06:
the sound is a wee bit stuttery in YT video's in v12, the video itself plays fine, the audio stutters gets worse with higher loads.

for example, the full version of minecraft 1.14.4 sounds and music becomes completely garbled while the game itself is playable.

this an extreme example though.

TheRemote, 2019-10-06:
Very interesting, I know people have reported sound issues lately in the kernel issues reports such as https://github.com/raspberrypi/linux/issues/3204

I also wonder if this happens on a 3B+ using the official image (which would point to more of an Ubuntu problem than a kernel problem).  We've already fixed several things that never worked properly in the official image (I think soon this image will be faster and more stable than the official one for Pi 2/3 at this rate).

I will investigate and see what can be done to fix the audio!

MadWomen, 2019-10-06:
this image is already faster than the official raspbian, i can actually play YT with this without video stutter, except for the audio.

i'm surprised i can actually play the latest version of minecraft.

good luck and thank you for bringig this distro to us.

johnjelinek, 2019-10-06:
Just a quick contrast: in official Raspbian, I get no garbled sound with this same use case.

johnjelinek, 2019-10-06:
Also, do you know how to make the RPi stop waiting for a network from the Ethernet on boot? I have a WiFi connection, so I want it to use that instead of waiting 2 minutes for an IP from eth0 and then timeout and then continue booting.

TheRemote, 2019-10-06:
Ahhh yes, that is thanks to cloud-init which I have honestly considered removing from the image entirely but it does a few critical things during the first boot of the Pi (it resizes the partition to use all the space on your SD card, generates SSH keys and enables SSH, etc).

On a fully set up Pi you can do sudo apt remove cloud-init and remove it.  If you try this make sure you have a monitor you can connect to in case your networking won't come up and you can SSH in but I tested this on my Pi in kubuntu-desktop and had no trouble getting back in after rebooting without the cloud junk.

johnjelinek, 2019-10-06:
Aight, I'll try that. But also, every time it boots, I get an error message:
> System program problem detected

There are buttons for
> Cancel
> Report problem...

johnjelinek, 2019-10-06:
Another thing: my wireless mouse has some input lag. I haven't noticed this lag from the wireless keyboard input. Both devices communicate to the same USB device. Any idea what would cause the mouse lag? I thought it was video-related, but if I drop my resolution to 1024x768, I get the same experience.

TheRemote, 2019-10-06:
I saw this once in testing in xubuntu-desktop (doesn't seem to happen in kubuntu-desktop).

As a quick test try:

sudo rm -f /var/crash/*

And reboot and see if the crash comes back.  If it does it may be a startup service issue in xubuntu-desktop (likely some config file somewhere) that we can iron out.

johnjelinek, 2019-10-06:
Note: the mouse lag is present on Raspbian too.

johnjelinek, 2019-10-06:
Another weird thing: I formatted the sdcard with v12, and installed ubuntu-mate-desktop while connected to one monitor. I turned it off and plugged it into another monitor and it complained about gpu-manager command. Reinstalling ubuntu-mate-desktop fixed it.

johnjelinek, 2019-10-06:
I removed cloud-init, but it still says on boot:
> A start job is running for Wait for Network to be Configured (... / no limit)

TheRemote, 2019-10-06:
The mouse lag one is interesting.  I use a Logitech K270 / M185 combo set that goes through the same USB adapter on mine and I haven't experienced this.  It could be a USB driver level issue.

One thing worth a try (but is a bit of a long shot) is try plugging your adapter into one of the other types of USB ports as a test.  If it's in a USB 2.0 one plug it into one of the blue USB 3.0 ones or vice versa and see if there is any change.

johnjelinek, 2019-10-06:
I plugged it into USB3 ports and same experience. Clicks are immediate just like typing, but movement is still delayed. It's almost like it eases into the mouse movement, with a slow start and finish, but it's fast in the middle of the movement (but still behind when I actually moved it).

Regarding the sound in v12, it's still garbled when I run `say hello`. Logging out and back in resolves that.

johnjelinek, 2019-10-06:
Note: if I put the same keyboard/mouse combo into Bluetooth mode, the delay goes away.

johnjelinek, 2019-10-06:
I plugged in USB headphones and the audio is no longer garbled. If I unplug, it becomes garbled again until I logout/reboot.

TheRemote, 2019-10-07:
I have been playing a video for quite some time without garbled audio so far with the following change.

`sudo nano /boot/firmware/cmdline.txt`

Remove what is in there and change your file to:

`snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_compat_alsa=0 dwc_otg.fiq_fix_enable=2 console=serial0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait rootflags=noload net.ifnames=0 `

Now reboot and give it some more tests.  I was using a HDMI monitor, but before this change I could hear artifacts starting almost immediately, and after I have not been able to get it to break.  I haven't tested with headphones yet, but this change should give you 2 sound devices instead of one

MadWomen, 2019-10-07:
still the same, it stables out but as soon as you start doing anything else, it start again, minecraft is also better but still a garbled mess.

TheRemote, 2019-10-07:
Are you using sound over HDMI or the headphone jack/speakers?  

Let's also try adding these lines to /boot/firmware/config.txt as another test:

# forces HDMI mode
hdmi_drive=2

# Pretends all audio formats are supported by display, allowing passthrough of DTS/AC even when not reported as supported.
hdmi_force_edid_audio=1

MadWomen, 2019-10-07:
i will try that, i've been using HDMI audio.

MadWomen, 2019-10-07:
no change sadly, yt works fine if you leave it be but as soon as you start scrolling or do some other tasks in the background, it starts to stutter and garble again.

TheRemote, 2019-10-08:
Just out of curiosity which method are you running Minecraft 1.14.4 with?  Obviously the official launcher doesn't work, and I tried a few crappy ones that didn't really work either.  I'd like to test in here since it sounds like it reproduces the problem really well!

MadWomen, 2019-10-08:
i've used this guide:
https://www.raspberrypi.org/forums/viewtopic.php?t=249875

however, you have to update the legacy launcher to 1.6.93 which you can get here:
https://launcher.mojang.com/v1/objects/eabbff5ff8e21250e33670924a0c5e38f47c840b/launcher.jar

when you have to edit the launch script, you have to edit the user path that reflects ubuntu and not the standard user of genthoo where this guide is based on.

pseudofunizer, 2019-10-11:
Crackling audio, especially when playing back video from a browser, has been a persistent problem in all Raspbian builds from aeons, the solution to that has been around for aeons as well, and it's plain stupid that it's not been implemented yet. It has to do with PulseAudio's scheduling. **SOLUTION:**

Edit the `/etc/pulse/default.pa` file, i.e.:

`sudo nano /etc/pulse/default.pa`

From the file, look for: 
`load-module module-udev-detect`

Add `tsched=0` to the end of that line, like so:

`load-module module-udev-detect tsched=0`

Save, exit, reboot just in case. Problem solved. That should fix almost all audio crackling issues in browsers, video playback, et cetera.

MadWomen, 2019-10-11:
thanks a bunch, that did the trick, no more garbled audio even under heavy load!

TheRemote, 2019-10-23:
This has been added in v13, thanks!

closed 2019-10-23 by TheRemote