TheRemote/Ubuntu-Server-raspi4-unofficial #46: V11 running pretty smoothly

antonical, 2019-10-06:
Just wanted to feed back that V11 is running pretty smoothly. I wanted to configure to be the same as another desktop machine.

Installed standard ubuntu-gnome-desktop all went well. I am sure that more tweaking can occur. 

I  just saw V12 is there a way to simply upgrade the firmware in V11 as I am in the testing phase.

Cheers
Spart

TheRemote, 2019-10-06:
Not as simple as before but still not too difficult.  The easiest way right now until I complete the update script would be to download the latest image and extract the /lib/firmware folder from there and copy it over the existing /lib/firmware from v11.

Then to implement the fixes I made in v12 you would do the following commands (on the Pi you want to update from inside Ubuntu):

sudo apt-get install bluez -y

# % Startup tweaks to fix bluetooth
sudo rm /etc/rc.local
cat << EOF | sudo tee /etc/rc.local
#!/bin/sh -e
#
# rc.local
#

# Enable bluetooth
if [ -n "`which hciattach`" ]; then
  echo "Attaching Bluetooth controller ..."
  hciattach /dev/ttyAMA0 bcm43xx 921600
fi

exit 0
EOF

sudo chmod +x /etc/rc.local

# % Add symbolic link from /etc/firmware to /lib/firmware (fixes Bluetooth)
sudo ln -s /lib/firmware /etc/firmware

Now reboot and you should have working Bluetooth as well as the other firmware updates implemented into /lib/firmware for v12

The update script I'm working on for exiting pre-release will essentially work this way.  I will commit the current files to this repository in folders like kernel-modules, firmware-nonfree, etc. and the script will copy these over and run the fixes from the various releases and make this much easier.

TheRemote, 2019-10-06:
Oops, forgot one step, you also need to change /boot/firmware/cmdline.txt to the following:

dwc_otg.fiq_fix_enable=2 console=serial0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait rootflags=noload net.ifnames=0

antonical, 2019-10-06:
Many thanks James.

Cheers
Spart

closed 2019-10-07 by TheRemote