TheRemote/Ubuntu-Server-raspi4-unofficial #47: Networking Problems

tgwaste, 2019-10-07:
Any tips on getting wifi working ?  I setup my netplan file just like any other raspberry pi running ubuntu and I see wlan0 in 'ip a' but it never connects to anything.

TheRemote, 2019-10-07:
Give it a:
sudo ifconfig wlan0 up 
command.  It seems like this only needs to be done once after installing and then it stays on unless you shut it off again

tgwaste, 2019-10-07:
Nadda :(

[image]

tgwaste, 2019-10-07:
FYI, I tried the normal raspbian image for the board and Wifi came right up with no problems.

bruces22, 2019-10-07:
Just finished install v12, installed xubuntu-desktop. Wlan0 set up OK using Settings,Network connection and add a wifi.

tgwaste, 2019-10-07:
v12 is what I’m using but I’m headless.  I’ve pretty much tried everything at this point. 

All I get is:
[  690.863431] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  690.864408] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  690.864419] brcmfmac: power management disabled

bruces22, 2019-10-07:
Maybe try iwconfig???? Radio on???

tgwaste, 2019-10-07:
Tried it. With both power on/off and Nadda

wlan0     IEEE 802.11  ESSID:off/any
          Mode:Managed  Access Point: Not-Associated   Tx-Power=31 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off

mateuszwasowski, 2019-10-07:
I had to add `dhcp4: true` under `wlan0`

mateuszwasowski, 2019-10-07:
🤔 have you tried removing `optional: true`? of course after changes run `sudo netplan generate`&& `sudo netplan apply`

tgwaste, 2019-10-08:
no difference after removing optional: true
same thing as above
[ 690.864408] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[ 690.864419] brcmfmac: power management disabled

TheRemote, 2019-10-08:
On a fresh install of just plain Ubuntu Server v12 I did the following:

```
sudo apt install wireless-tools

sudo nano /etc/wpa_supplicant.conf

network={
        ssid="fakessid"
        psk="fakepassword"
}

sudo wpa_supplicant -B -c /etc/wpa_supplicant.conf -i wlan0

ifconfig wlan0 up

sudo dhclient wlan0

ifconfig

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.88.130  netmask 255.255.255.0  broadcast 192.168.88.255
        inet6 fe80::dea6:32ff:fe02:8cea  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:02:8c:ea  txqueuelen 1000  (Ethernet)
        RX packets 736  bytes 191806 (191.8 KB)
        RX errors 0  dropped 70  overruns 0  frame 0
        TX packets 17  bytes 2338 (2.3 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

iwconfig

wlan0     IEEE 802.11  ESSID:"fakessid"
          Mode:Managed  Frequency:2.442 GHz  Access Point: CC:2D:E0:A9:D5:0F
          Bit Rate=135 Mb/s   Tx-Power=31 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
          Link Quality=70/70  Signal level=-9 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

dmesg

[  152.659826] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  152.659844] brcmfmac: power management disabled
[  456.584356] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
```

TheRemote, 2019-10-08:
My guess is you're dealing with a netplan configuration issue and that using the above commands will work unless something *really* strange is going on.

I noticed that in your .yaml file you were using for netplan that your wlan0 is way more indented/spaced than your eth0 entry.  These .yaml configuration files completely break if there's even one space that doesn't belong in there which is why I despise them.  I'd paste your contents into a yaml validator like https://codebeautify.org/yaml-validator and I don't see how it could pass looking at the screenshot.

Hopefully this helps get things on the right track!

tgwaste, 2019-10-08:
Following steps above, it hung on: dhclient wlan0 however I tried my 2.4ghz wifi instead and it works:

```
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.0.1.51  netmask 255.255.255.0  broadcast 10.0.1.255
        inet6 fe80::dea6:32ff:fe1b:ddf7  prefixlen 64  scopeid 0x20<link>
        ether dc:a6:32:1b:dd:f7  txqueuelen 1000  (Ethernet)
        RX packets 11  bytes 2126 (2.1 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 11  bytes 1782 (1.7 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```
My 5GHz works on Raspbian so must be something in Ubuntu.

tgwaste, 2019-10-08:
reboot breaks this.  how to make permanent.

tgwaste, 2019-10-08:
It's still not really working.
Even on reboot I see wpa_supplicant running but when I do dhclient wlan0 it just hangs forever.
Got it to work once but I can not repeat the success.
I wonder if the wifi cards are not the same between Pis.

As a side: I also notice in dmesg I get a bunch of these errors:
[image]
Ive installed this on about 6 different SD cards of various types and sizes and it always happens.

TheRemote, 2019-10-08:
Those commands were just diagnostic to test the card without relying on any configuration.  You need to fix your netplan configuration if you want everything applied nicely by Ubuntu in the stock image as-is.  There's a ton of alternative ways to configure the network as well like using Network-Manager or several other network management tools.

5Ghz is enabled by setting your CRDA country code to match where you are.

sudo apt-get install iw

Then setting you regulatory database in sudo nano /etc/default/crda to your home country (I used US).  After a reboot:

wlan0     IEEE 802.11  ESSID:"fakessid5G"
          Mode:Managed  Frequency:5.18 GHz  Access Point: CC:2D:E0:A9:D5:0E
          Bit Rate=390 Mb/s   Tx-Power=31 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
          Link Quality=70/70  Signal level=-24 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

This is a type of change I could integrate into this image for convenience because the Pi did not have 5Ghz support when the official image was made so it is a nice enhancement but it's all just configuration.  

The image literally uses the latest official Raspbian release firmware, kernel and modules.  There's nothing magic in it that makes anything work.  There's no different drivers or firmware.  It's just configuration files and packages.

I can check into the fsck issue but it's really a separate one so I don't want to leave this open for that.

I'd definitely start here: https://www.idmworks.com/netplan-for-newbies/ to get a grasp on netplan because trying to work around the official Ubuntu way of doing it is going to cause nothing but headaches.

closed 2019-10-08 by TheRemote