TheRemote/Ubuntu-Server-raspi4-unofficial #48: How can I use the CSI Raspberry Pi Camera

jensJJ, 2019-10-07:
Thank you, I tested the newest build on the Rpi4 works good! I was trying to run the CSI camera, but cannot get it running using gstreamer, since it does not appear on video0 or any. Do you know how to fix this?

TheRemote, 2019-10-23:
It sounds like that camera uses i2c.  In v13 I added raspi-config in which should help you get everything enabled.  Can you try with the latest release?

jensJJ, 2019-10-23:
Hello, do I have to download the new build image or can I update dist on my current RPI4?

TheRemote, 2019-10-24:
Right now you would need to download and reinstall, but the update feature is coming soon (TM).

No, but seriously, now that the fsck errors and hardware errors are under control I want to get the update script out in the next release or two!

As for the camera issue, it should be as easy as going into raspi-config and turning on the camera interface/drivers just like you would in something like Raspbian Lite.  If this gives you any trouble definitely reopen this issue and we can troubleshoot it further.

closed 2019-10-24 by TheRemote