TheRemote/Ubuntu-Server-raspi4-unofficial #51: How to use cloud-init

dkrizic, 2019-10-10:
Hi,

just a simple question. I saw that Ubuntu has cloud init installed. I would like to configure things like:

- hostname
- repositories
- software installation
- users
- rsa-keys

How do I use it? Is it sufficient to put a cloud.yaml into /boot after flashing the image?

dkrizic, 2019-10-11:
The guys at Hypriot overwrite the cloud.cfg with this file:

https://github.com/hypriot/image-builder-rpi/blob/master/builder/files/etc/cloud/cloud.cfg so it will not try to contact any external servers. This would enable keeping cloud-init without delays on boot. I am currently investigating how they manage to read the configuration files from /boot

damarshall, 2019-10-11:
@dkrizic you can also bootstrap your cluster using the nocloud method for `cloud-init`. I've documented the process [on this page](https://github.com/damarshall/rpi4utils/blob/master/docsite/content/docs/cloud-init/index.md). In essence, you create a *cidata* volume (type vfat) on a small external USB drive and insert that only for first boot. I use this method to bootstrap pi4 clusters.

Here's a link to a [directory with template cloud-init content](https://github.com/damarshall/rpi4utils/tree/master/cloud/seed).

You'll have to edit `meta-data` and `network-config` between boots of subsequent nodes in your cluster, and you have to bootstrap a node at a time but I've done it several times and can bring up a cluster of 8 nodes in about an hour, including imaging drives.

dkrizic, 2019-10-11:
Nice one @damarshall , this sounds like a nice alternative approach!

closed 2019-10-11 by dkrizic