TheRemote/Ubuntu-Server-raspi4-unofficial #55: Fail to set_network in Pre-v12

KrisWangzZ, 2019-10-20:
When configuring the WiFi ssid:  `wpa_cli -i wlan0 set_network 0 ssid "DESKTOP"`, I got a `FAIL`.
So I trace the command, and get the following:
```
execve("/sbin/wpa_cli", ["wpa_cli", "-i", "wlan0", "set_network", "0", "ssid", "DESKTOP"], 0x7fcb245d30 /* 21 vars */) = 0
brk(NULL)                               = 0x556ea82000
faccessat(AT_FDCWD, "/etc/ld.so.nohwcap", F_OK) = -1 ENOENT (No such file or directory)
faccessat(AT_FDCWD, "/etc/ld.so.preload", R_OK) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=46339, ...}) = 0
mmap(NULL, 46339, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7fb0749000
close(3)                                = 0
faccessat(AT_FDCWD, "/etc/ld.so.nohwcap", F_OK) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/aarch64-linux-gnu/libreadline.so.7", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\300;\1\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=282088, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fb0729000
mmap(NULL, 351200, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fb06d3000
mprotect(0x7fb0710000, 65536, PROT_NONE) = 0
mmap(0x7fb0720000, 32768, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x3d000) = 0x7fb0720000
mmap(0x7fb0728000, 3040, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fb0728000
close(3)                                = 0
faccessat(AT_FDCWD, "/etc/ld.so.nohwcap", F_OK) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/aarch64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0 \10\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=1341080, ...}) = 0
mmap(NULL, 1409880, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fb057a000
mprotect(0x7fb06ba000, 61440, PROT_NONE) = 0
mmap(0x7fb06c9000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x13f000) = 0x7fb06c9000
mmap(0x7fb06cf000, 13144, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7fb06cf000
close(3)                                = 0
faccessat(AT_FDCWD, "/etc/ld.so.nohwcap", F_OK) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/lib/aarch64-linux-gnu/libtinfo.so.5", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0\267\0\1\0\0\0\0\312\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=162432, ...}) = 0
mmap(NULL, 228008, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7fb0542000
mprotect(0x7fb0566000, 61440, PROT_NONE) = 0
mmap(0x7fb0575000, 20480, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x23000) = 0x7fb0575000
close(3)                                = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fb0540000
mprotect(0x7fb06c9000, 16384, PROT_READ) = 0
mprotect(0x7fb0575000, 16384, PROT_READ) = 0
mprotect(0x7fb0720000, 8192, PROT_READ) = 0
mprotect(0x555db8f000, 12288, PROT_READ) = 0
mprotect(0x7fb0757000, 4096, PROT_READ) = 0
munmap(0x7fb0749000, 46339)             = 0
brk(NULL)                               = 0x556ea82000
brk(0x556eaa3000)                       = 0x556eaa3000
rt_sigaction(SIGINT, {sa_handler=0x555db76580, sa_mask=[INT], sa_flags=SA_RESTART}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 8) = 0
rt_sigaction(SIGTERM, {sa_handler=0x555db76580, sa_mask=[TERM], sa_flags=SA_RESTART}, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 8) = 0
socket(AF_UNIX, SOCK_DGRAM, 0)          = 3
getpid()                                = 2550
bind(3, {sa_family=AF_UNIX, sun_path="/tmp/wpa_ctrl_2550-1"}, 110) = 0
connect(3, {sa_family=AF_UNIX, sun_path="/var/run/wpa_supplicant/wlan0"}, 110) = 0
fcntl(3, F_GETFL)                       = 0x2 (flags O_RDWR)
fcntl(3, F_SETFL, O_RDWR|O_NONBLOCK)    = 0
sendto(3, "SET_NETWORK 0 ssid DESKTOP", 26, 0, NULL, 0) = 26
pselect6(4, [3], NULL, NULL, {tv_sec=10, tv_nsec=0}, NULL) = 1 (in [3], left {tv_sec=9, tv_nsec=999997481})
recvfrom(3, "FAIL\n", 4095, 0, NULL, NULL) = 5
fstat(1, {st_mode=S_IFCHR|0620, st_rdev=makedev(136, 0), ...}) = 0
write(1, "FAIL\n", 5FAIL
)                   = 5
clock_gettime(CLOCK_BOOTTIME, {tv_sec=11224, tv_nsec=324316235}) = 0
unlinkat(AT_FDCWD, "/tmp/wpa_ctrl_2550-1", 0) = 0
close(3)                                = 0
exit_group(0)                           = ?
+++ exited with 0 +++
```
What shoud I do to fix the problem?

TheRemote, 2019-10-23:
Can you try with v13?  There is a lot more wireless capability built in now

TheRemote, 2019-10-24:
Let me know if you try the new version and are having issues but a ton of work was done on the wireless stack to give it 5 GHz support so it's a completely different set of firmware now.

closed 2019-10-24 by TheRemote