TheRemote/Ubuntu-Server-raspi4-unofficial #58: Official image released

antonsamper, 2019-10-23:
I’ve just found  the following page which suggests Ubuntu Server is now officially supported on Rasp Pi 4 (64bit)

https://ubuntu.com/download/iot/raspberry-pi

Have you had a chance to try it? Does it differ from your image?

TheRemote, 2019-10-24:
I have not tried it yet but someone commented on my site saying it was out.  They said they still had some problems with the USB working on it.

It's also Ubuntu 19.10 so it only has a 9 month release cycle before it's obsolete since it isn't a LTS release like 18.04.

I'm downloading the image right now and going to give it a go!  I'm excited to see what tricks they may have used and how stable the support is going to be.  I'll definitely report back!

rmandrad, 2019-10-24:
> I have not tried it yet but someone commented on my site saying it was out. They said they still had some problems with the USB working on it.
>
> It's also Ubuntu 19.10 so it only has a 9 month [...]

quick question - so I upgraded your version to 19.10 however how to enable the new 5.3 kernel? ... I unmarked the raspi2 packages and flash-kernel ... the only way I can get a new kernel enabled is by compiling it. Thank you

peardox, 2019-10-24:
Maybe they'll fix the USB for 20.04 LTS a year after Pi4 release?

TheRemote, 2019-10-24:
The USB is a total mess.  If you do a sudo dmesg you'll see a huge stream of USB descriptor errors in there.  I tried several keyboards / mice in both the 2.0 and 3.0 ports and couldn't get anything out of them.

I also installed kubuntu-desktop as a test and after a boot the system logs are full of errors and warnings (both hardware and software).

My final verdict: needs some fixes before it's ready for prime time.  I'll keep maintaining and improving mine for the near future as well as we haven't heard any news as to whether raspi4 will be backported to the 18.04 LTS branch since 19.10 support ends in 2020 which makes it not the best option for everyone.

I feel like the image is just barely coming together right as an official release drops!  I'm working on integrating the kernel headers to build DMKS modules and allowing people to update the image via an apt ppa repository.

If you guys find anything that is working better in 19.10 than it does in mine definitely let me know so I can see what sneaky tricks they did to get it to work!

TheRemote, 2019-10-27:
This will still be supported for the near future.  There's still one thing I need to do which is finish the ability to update inside the image and then I will do the "official" release.

I learned countless things doing this project and if I attempt something like this again in the future I'll be a lot more ready for it!

closed 2019-10-27 by TheRemote

stephenjamieson, 2019-11-01:
@TheRemote just curious, do you have a list of steps that would be needed to upgrade to 19.10.  Reading through the script but I don't want to miss anything.