TheRemote/Ubuntu-Server-raspi4-unofficial #59: Release Updater

bruces22, 2019-10-28:
Issue with Updater.sh.  Tried on V14, attached pic for details.
(Also still trying Boot with ssd)
[image]

six4m, 2019-10-28:
Same Problem here !

TheRemote, 2019-10-28:
I'm a dummy and put a code grouping in the wrong place!

Can you both redownload and try again?

tikiatsbme, 2019-10-28:
Currently running Raspbian 32Bit as it works.

Here the /var/log/syslog file contents when it stopped working. 

[syslog.logfile.txt](https://github.com/TheRemote/Ubuntu-Server-raspi4-unofficial/files/3780719/syslog.logfile.txt)

[syslog.logfile.txt](https://github.com/TheRemote/Ubuntu-Server-raspi4-unofficial/files/3780726/syslog.logfile.txt)

six4m, 2019-10-28:
now its working, thanks and all thumbs up your great work  :)

closed 2019-10-28 by TheRemote