TheRemote/Ubuntu-Server-raspi4-unofficial #60: Raspberry Pi 4B_4GB failed install with release V15 Public

tikiatsbme, 2019-10-28:
I xzcat the V15 XZ file, then I DD burned the V15 IMG file to a 32GB SanDisk Ultra. Shut down to RPi4B, inserted to micro sd card and turned it on. It booted for about 47.11 seconds until it got 3 lines pass "----END SSH HOST KEYS ---" and stopped dead in its tracks with flashing "_" underline symbol. 

No response from keyboard. Tried Ctrl-C and Return/Enter with no success. Had to shutoff pi and pull card which is on my desk. Any log file I should upload for trouble shooting??

I tried this with the stock Ubuntu 19.10 Arm64 and got similar results which itself is interesting, possible unrelated.

TheRemote, 2019-10-28:
/var/log/syslog would be good

The SSH keys part is a part of the cloud-init stuff but it doesn't necessarily mean it is what is locking it up.

Does Raspbian boot on it?

tikiatsbme, 2019-10-28:
,Using raspian 32Bit right now

Thomas J Killeen, PE (MA & NH), ME RMS #351
I-Phone: 207-752-4102
thomas.killeen@icloud.com
46 Woodland Hills
South Berwick, ME 03908-1834
Sent from iCloud

tikiatsbme, 2019-10-28:
Currently running Raspbian 32Bit as it works.

Here the /var/log/syslog file contents when it stopped working.

syslog.logfile.txt

TheRemote, 2019-10-28:
Can you try unplugging all of those USB devices and booting for a test?

tikiatsbme, 2019-10-28:
I tried it with all USB devices removed, that is keyboard, mouse, usb hub and usb flashdrive removed, got as far as login that required use of keyboard. Plugged in USB keyboard (mac 108 kybd) - not recognized. 

[syslog.logfile.txt](https://github.com/TheRemote/Ubuntu-Server-raspi4-unofficial/files/3781004/syslog.logfile.txt)

boxlady, 2019-10-29:
Might be this? 
https://bugs.launchpad.net/ubuntu/+source/linux-raspi2/+bug/1848703
Try putting this in the config.txt
total_mem=3072

TheRemote, 2019-11-01:
This should be working again in V16, let us know if you're still having issues!

closed 2019-11-01 by TheRemote