TheRemote/Ubuntu-Server-raspi4-unofficial #61: SD CARD COPIER

tikiatsbme, 2019-10-29:
Is "SD CARD COPIER" provided with Ubuntu-Server-raspi4-unoffical. 

If yes, great and closed. 

If no, we need a GUI way to easy make a clone of a working micro SD card. Not sure if "env", "Sudo=_ASKPASS", "Usr/lib/piclone/pwdpic.sh", "usr/bin/piclone" even exist in an Ubuntu server image or Ubuntu desktop image. I doubt that Ubuntu would incorporate such, so its needs to be added somehow. ...Just Wondering......

TheRemote, 2019-10-31:
I had never heard of this utility before.  It looks like it's a Raspbian app.  I do provide builds of several Raspbian apps (like vcgencmd, raspi-config) in the image already.  I can tell you for certain that this specific utility doesn't exist anywhere else (it even has PiClone in the name).

I'm not quite sure what to make of the second half of your post, but I think you're trying to tell me the pre-reqs needed to build the application or you're describing the process you went trying to build it yourself.  

Either way you're way overthinking it here.  They provide everything they need on their site to build this application in < 10 minutes, which is why everyone over there is treating you like crap.

```
sudo apt install git build-essential libgtk2.0-dev
git clone https://github.com/raspberrypi-ui/piclone.git
cd piclone
./autogen.sh
dpkg-buildpackage -us -uc
```

Here I got some errors saying I needed libgtk2.0-dev, so I did apt-get install libgtk2.0-dev.  Continued:
```
dpkg-buildpackage -us -uc
Build succeeds.  If I do a:
cd ..
ls
```
I now see:

```
piclone
piclone_0.15_amd64.buildinfo
piclone_0.15_amd64.changes
piclone_0.15_amd64.deb
piclone_0.15.dsc
piclone_0.15.tar.xz
piclone-dbgsym_0.15_amd64.ddeb
```

Now I have a .deb package that I can take and install anywhere I want.  I don't care where it puts the files or what is working under the hood (although you can open the .deb in Archive Manager and look where everything goes).  This is why I said you're overthinking it.  It's a tiny and very simple program.

I want to test it now, so I'm going to do:

`sudo dpkg -i piclone_0.15_amd64.deb`

Which gives me:

```
Selecting previously unselected package piclone.
(Reading database ... 192765 files and directories currently installed.)
Preparing to unpack piclone_0.15_amd64.deb ...
Unpacking piclone (0.15) ...
dpkg: dependency problems prevent configuration of piclone:
 piclone depends on uuid; however:
  Package uuid is not installed.
```
Okay, so I got another error when trying to install saying I need the uuid package.  By now you know what to do:

`sudo apt-get install uuid`

Okay now I try again and it succeeds.  The application is now installed.  

[image]

I actually built this in x86_64 (amd64) using a Ubuntu 18.04.3 virtual machine.  Since I built it on x86_64 and everything is working we can easily assume that the ARM64 build will be even easier.

Give it a try.  This took less than 10 minutes for me, and even if it takes you a couple of hours I guarantee you it will take you far less time to do the build I just walked you through than it takes to go around trying to convince everyone to build it for you or integrate it into their image.

It wasn't hard to build.  What you're trying to do though is HARD.  You're trying to convince other people to build for your specific use case without even trying to do it yourself.  

If you had tried to build first and went to the piclone guys with a dependency or build error they would have treated you with much less hostility and probably even helped you get to where you needed to go as long as you were putting in the effort and trying to learn.  The Linux community as a whole is actually way less nice than the Pi community too!

I wrote this though because I am hoping it helps.  When everyone starts out they don't even know what questions they should be asking or where to start.  Start by building and the rest will fall into place!

tikiatsbme, 2019-11-01:
Thank you, so I put everything in a directory and follow your CLEAR instructions, yes! Thank you!!!!!

Thomas J Killeen
207-752-4102 (Cell)
46 Woodland Hills
South Berwick, ME 03908-1834
Thomas.killeen@icloud.com
Tikiatybme@gmail.com

TheRemote, 2019-11-01:
No problem, glad it helped!

closed 2019-11-01 by TheRemote