TheRemote/Ubuntu-Server-raspi4-unofficial #62: No sound (at least from analog out) after V15 update

pseudofunizer, 2019-10-29:
Hi there! Thanks a lot for the otherwise great port for the Ubuntu Server, I've been using it for some months now and it's great!

However, just today as I updated to v15, I think something went wrong with the audio drivers. 

Although the analog audio output still gives the "on/off crackle" when initializing, I can no longer hear anything from the analog audio output. Upon probing, I noticed that Pulse Audio Mixer shows only the Dummy Audio Output as the only available device. However, as for ALSA, alsamixer seems to find both the headphone out and the HDMI out. 

I tried some of the usual tricks like pulseaudio -k , and then reloading ALSA, also tried to purge both alsa-base and pulseaudio, and reinstall them, etc.

with commands like `sudo alsa force-reload`, nothing happened, apart from that "promising" initial "snap-crackle" that gives an indication that the analog output is ready.

I've tried the usual troubleshooting to find out what's listed in there, for instance:

	`cat /proc/asound/cards`

lists both the HDMI ( instance 0 ) and the headphone output ( instance 1 ), and I get the crackle with sudo alsa force-reload indicating that at least its probing the analog audio output, however `speaker-test` gives out no sound, tried setting the default for headphones with the official ALSA instructions. https://www.alsa-project.org/wiki/Setting_the_default_device 

The irony here is that I'm the guy who helped out in the blog comments of this distro fork on how to configure PulseAudio so that it doesn't crackle up when playing back web content, and now the audio output at my end is totally kaput (at least for the analog audio output). D'oh! Oh well, you win some, you lose some...

pseudofunizer, 2019-10-29:
I just noticed that there was a firmware update ( `rpi-update` ) released just 17 hours ago related to some audio issues ( https://github.com/Hexxeh/rpi-firmware/commit/c5736330216628b5ff8e3d17dde7cc03ce2126e6 ), will have to see if I had it installed already or not.

TheRemote, 2019-10-29:
Hey pseudofunizer,

I remember your sound posts, thanks for those!

So I have some ideas.  The most likely thing to test is to change your /boot/firmware/cmdline.txt flags.  This was added not too long ago and may have changed the behavior:

snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_compat_alsa=0 

These 3 kernel flags are flags I added but honestly I don't think it made any difference other than it lets you use both sound devices at once and control them separately.

I doubt that sound patch you just posted about is in there either!  I'll do a v16 build tonight to get all the changes that are being pushed through as well, but in the mean time try changing/removing those kernel flags and rebooting just to help us keep narrowing down the behavior!

pseudofunizer, 2019-10-29:
> I doubt that sound patch you just posted about is in there either! I'll do a v16 build tonight to get all the changes that are being pushed through as well, but in the mean time try changing/removing those kernel flags and rebooting just to help us keep narrowing down the behavior!

Thanks for those tips, Capt'n! I'll try them out and will keep you posted. Cheers!

TheRemote, 2019-10-29:
No problem!  There's also a lot of changes in the git changelog you sent me that we are going to want so I'm going to start building v16 in case the flags don't help!

pseudofunizer, 2019-10-29:
I stared blankly at the wall for a while there, and came to the conclusion that since something probably broke before this update (and in any case, I'm the one that caused it to break due to user action, hehe), the safest bet would be for me to revert to one of my backup images and rebuild from there on. 

Especially the "now it's working, now it's not"-juggling that I'm experiencing is a bit too unnerving for me -- I also might have installed some stuff in between that might've affected the update in of itself. 

Perhaps related, perhaps not: there's something strange in running the Ubuntu Server from an external SSD w/ the jump-boot from MicroSD, even if the external drive is properly powered up from a separate PSU. I've now had two instances of a brand-new WD Blue SSD going to sh*t in terms of the filesystem (ext4) after a few weeks of use. Funnily enough, I haven't had any of those problems with Raspbian, and I've had an SSD attached to that build with a USB-SATA adapter that doesn't even have an external power supply! 

Oh well. The torture never stops. Hehe.

TheRemote, 2019-10-29:
I think it was almost certainly the updates!  Not to worry though, we have every release img backed up in the downloads section.  The absolute worst case scenario is that the v16 build doesn't fix it and I start digging for which firmware file or module broke it.

Changing out sets of firmware and even kernel modules is pretty simple.  If you still have your original download (or know what version it is) you could even copy the /lib/firmware folder onto your existing device and reboot.  If it ends up being a kernel issue it's about the same but with a few extra steps.

We're definitely on the bleeding edge right now with the Pi 4.  I'm pretty shocked that the USB boot support (a requirement in my opinion) is STILL not done!  Also if you look at those changelogs it's pretty amazing what they are still in the middle of implementing.  This release definitely felt pushed out a little early.

Let's test from v16 and go from there

reddragonemperor69, 2019-10-30:
I'm having the same issues with audio,works for the 1st boot but after reboot(either the 1st or the 2nd),it doesn't work,no audio output is shown,i have tried releases. v15,v14 and v13 all have the same issues.I have updated to the latest firmware and latest bootloader,also tried playing around with cmdline parameters that was mentioned and still no avail,other than that everything seems to be working fine.

I have the 4gb Raspberrypi4B currently on the v13 build with Kubuntu Desktop.

TheRemote, 2019-10-31:
Hello,

I found some problems.  Your post helped quite a bit reddragon as I was able to reproduce it breaking after the first restart and fix it!

There's also some kernel flag changes to help the sound.  One kernel flag related to memory allocated for the onboard sound was allocating as 16 instead of 2048 for example, pretty big difference!

You would have also been affected pseudo, definitely give the new build a try!

pseudofunizer, 2019-11-01:
OK, making some progress here -- at least in terms of bug-hunting! :P 

The Updater.sh script went haywire during update due to the fact that I had already begun to use my SSD as the main drive (FAT bootfs from MicroSD, system on SSD over a SATA-to-USB adapter.)

My `/etc/fstab` had been updated according to those changes, and the `crypttools' portion of the Updater script could not handle that -- it couldn't find the root file system. Oh well... At least the update procedure did acknowledge that it couldn't create some of the required installations due to that fact, so it saved me a lot of troubleshooting time. 

I would suggest adding a `--reinstall` or `--force-reinstall` command line switch option to the Updater script so that it would disregard the existing version number that is written at `/etc/currentrelease` and just piledrive on through with the process regardless of what it finds as a pre-existing installation. 

On the other hand, it might introduce some new problems, but as for me, now that I noticed that things started to go wrong with the /etc/fstab interpretation, I could backtrace the problem, modify the `/etc/currentrelease` file and redo the installation over the old one. 

Then again, should `crypttools` be unable to determine what it wants out of /etc/fstab, it all goes to hell at the next reboot.

In addition to the suggestion of having a forced reinstall, since something can and in many cases _will_ go wrong during updates (Ubuntu's own Update Manager is no exception to that rule), perhaps some amount of attention should be put into checking the `/etc/fstab` status before proceeding with the update process at all. 

I'll have to revert to yet another previous build image that I have (that doesn't have changes in the `/etc/fstab` from the original contents) and retry the Updater process.

reddragonemperor69, 2019-11-01:
@TheRemote  Hi! Thank You for the response and the amazing build,i have been enjoying it(except for the whole sound issue)  more then the official 19.10 build.

So, i tried the v16 update and it still didn't work,at first i tried through the updater script and it didn't work so i thought i would try the release image in case something may have went wrong with the updater script,and the same thing happened,first boot,sound was working,rebooted and it still worked so i thought i should reboot once more to confirm so i rebooted again and then it was gone,same thing as before.

BTW i noticed something will it was booting up and there was an error loading `systemd-load-modules` ,i don't know if this is related to it but i'll attach the journalctl output with it,hope it helps,really hoping to see the audio work so let me know if i can help with something else to figure this out.
[journactl.txt](https://github.com/TheRemote/Ubuntu-Server-raspi4-unofficial/files/3798866/journactl.txt)

pseudofunizer, 2019-11-01:
One more update on the troubleshooting: I just reverted to an earlier backup of a working image and the updates installed without problems APART from the PulseAudio driver, which, upon clicking on the speaker icon with an "X" next to it (= indicating that it's not functioning), says:

`Connection to PulseAudio failed. Retrying in 5s.

In this case this is likely because PULSE_SERVER in the Environment/X11 Root Window Properties or default-server in client.conf is misconfigured.

This situation can also arise when PulseAudio crashed and left stale details in the X11 Root Window. If this is the case, then PulseAudio should autospawn again, or if this is not configured you should run start-pulseaudio-x11 manually.` 

Wow. Oh well, back to troubleshooting!

TheRemote, 2019-11-01:
You are both saints for testing and giving such detailed feedback!

So I had done all my testing with kubuntu-desktop instead of xubuntu desktop.  I just installed it and when I run sudo systemctl list-units --failed I am seeing a LightDM failure and a systemd-load-modules failure.

Next I issued a sudo systemctl restart systemd-modules-load and then a sudo systemctl status systemd-modules-load

Nov 01 17:51:22 ubuntu systemd[1]: Starting Load Kernel Modules...
Nov 01 17:51:22 ubuntu systemd-modules-load[1529]: Failed to find module 'lp'
Nov 01 17:51:22 ubuntu systemd-modules-load[1529]: Failed to find module 'ppdev'
Nov 01 17:51:22 ubuntu systemd-modules-load[1529]: Failed to find module 'parport_pc'
Nov 01 17:51:22 ubuntu systemd[1]: systemd-modules-load.service: Main process exited, code=exited, status=1/FAILURE
Nov 01 17:51:22 ubuntu systemd[1]: systemd-modules-load.service: Failed with result 'exit-code'.
Nov 01 17:51:22 ubuntu systemd[1]: Failed to start Load Kernel Modules.

And I have seen these evil lines before!  parport_pc is try to load a 1990s style parallel port driver (which obviously the Pi does not have a parallel port).  I'll trace the source of it here.  You guys can run the same systemd commands to see which one of your services is failing.  Once we know that it's not too bad to figure out what is happening from there!

TheRemote, 2019-11-01:
Bonus command:  sudo journalctl -xe and start hitting page up until you see red text!

I think I am reproducing what you guys are seeing now that I'm trying with xubuntu desktop though

TheRemote, 2019-11-01:
Okay I'm making some progress, I'm able to fix the kernel modules not loading with

sudo rm /etc/modules-load.d/cups-filters.conf

Or you can go in there and comment the lines out (you'll recognize the 3 modules it's trying to load as the ones that were in journalctl)

Next on to the LightDM service!

TheRemote, 2019-11-01:
Alright, v17 is out, can you two give it a try?

pseudofunizer, 2019-11-01:
> 
> 
> Alright, v17 is out, can you two give it a try?

I'm on it! Will keep you posted.

pseudofunizer, 2019-11-01:
> 
> 
> Alright, v17 is out, can you two give it a try?

Just ran the update, and, well, if it ain't one of these moments at 01 AM my local time on a Fri-Sat night:

https://youtu.be/HUInWKK8ROc?t=34

In other (less insane) words: **_IT WORKS!!!_**

pseudofunizer, 2019-11-01:
> 
> 
> Alright, v17 is out, can you two give it a try?

... There's only one small glitch; the system stereo audio output now shows up as two separate "Built-in Audio Mono" sources. Running `speaker-test` also proves that all audio out is monophonic, hence there's something a bit off with the L/R channels getting to the driver (?) or driver handler(?) correctly. **sadtrombone**

reddragonemperor69, 2019-11-02:
> Alright, v17 is out, can you two give it a try?

So i updated it and **IT WORKS!!!** Thank you so much BUT now for some reason the samba share that i was connected to earlier refuses to connect now other then that everything works great.After installing smbclient and running `smbclient -L //My samba share` shows `Error NT_STATUS_HOST_UNREACHABLE` but i tried connecting from my main machine and from the sd card that has raspbian installed on my Pi4 it works fine.

I guess i'll try flashing the v17 image and have a clean start to see if it persists.

Anyway,once again thank you so much for the quick response and fixes.

pseudofunizer, 2019-11-02:
OK, I've been troubleshooting the dual mono issue today, @reddragonemperor69 , can you check from your audio options that your audio output is not mono / 2x mono (that cannot be switched to a single stereo by default) ? I have that problem, although the audio out works otherwise just fine. 

**I think I found the culprit and the solution to the 2x mono outputs instead of a single stereo. I think the 2x mono out-only in the current version (v17) is a problem in the configuration of Ubuntu's audio sinks:**

https://superuser.com/questions/657876/mono-sound-output-in-ubuntu

One important notion: I noticed that the other mono output channel was named **"mono"** and the other **"Mono"** (with a capital M) -- Random-Sherlocking here, but... case sensitivity, maybe?

reddragonemperor69, 2019-11-02:
So i went ahead with a clean v17 img and installing desktop(this time went with unity and it's working fabulously ) and everything seems to be working great for now,no issues with audio or connecting to samba share,will test it more and reboot it a few times to ensure that everything is okay.

@pseudofunizer I haven't looked into the mono output thing and  since i had a portable speaker connected to it so i haven't noticed anything will check it out and let you know.

pseudofunizer, 2019-11-02:
Been shuffling through `pacmd` and them internets; apparently the "mono only" output problem is quite common, especially with SBC's. 

See: https://ubuntu-mate.community/t/audio-output-mono-instead-of-stereo-ubuntu-18-04-1-rpi/18001

TheRemote, 2019-11-03:
Sounds likely, does it work correctly in 19.10?  Or is Raspbian the only known one?  It should just be a matter of digging through their files and seeing how they configured it

TheRemote, 2019-11-04:
I got it, the following will fix it:

```
sudo sed -i '/^\[Mapping analog-mono\]/,+1s/device-strings = hw\:\%f.*/device-strings = fake\:\%f/' /usr/share/pulseaudio/alsa-mixer/profile-sets/default.conf
pulseaudio -k
pulseaudio --start
```

What does it do?  It disables the analog mapping in Pulse Audio which forces it to search for other ways to map the device.  The only other mapping in that file that will match is stereo-fallback.

You now can adjust the left and right volume channels independently and if you do the "Left" and "Right" tests you can tell that it is really stereo!

This is a very old Debian trick (apparently) that I found in the Raspberry Pi forum here: https://www.raspberrypi.org/forums/viewtopic.php?t=251455#p1557580

This will be integrated into V19 but anyone who wants it early can run the above lines

closed 2019-11-04 by TheRemote