TheRemote/Ubuntu-Server-raspi4-unofficial #65: Netplan fix

DaftHonk, 2019-11-01:
Why does the netplan fix rewrite the entire .yaml file? I have my pi set with a static IP and each time I run the updater it resets my interface to DHCP. Would it be better to inject the optional line into the file using `sed`?

TheRemote, 2019-11-01:
Yes it would, sorry about that, the netplan fix is relatively new and I basically just copied the same one over from the build script.  I'll add some precision to the fix in Updater.sh to prevent clobbering any existing configuration.

TheRemote, 2019-11-01:
I've changed it to this:

# Fix netplan
GrepCheck=$(cat /etc/netplan/50-cloud-init.yaml | grep "optional: true")
if [ -z "$GrepCheck" ]; then
    sudo rm -f /etc/netplan/50-cloud-init.yaml
    sudo touch /etc/netplan/50-cloud-init.yaml
    cat << EOF | sudo tee /etc/netplan/50-cloud-init.yaml
    network:
        ethernets:
            eth0:
                dhcp4: true
                optional: true
        version: 2
EOF
    sudo netplan generate 
    sudo netplan --debug apply
fi

So basically if you already have an optional: true in the netplan file (meaning the fix has already been applied) it leaves it completely alone

closed 2019-11-01 by TheRemote