TheRemote/Ubuntu-Server-raspi4-unofficial #67: OpenGL 3.2

peardox, 2019-11-02:
OBS-Studio potentially runs on a Pi4 but it requires OpenGL 3.2. 
It should be noted that the Jetson Nano has 3.2 but has some unknown bug I'm trying to track down - possibly something in SSE (I'm using sse2neon.h for that but one macro is missing)

Is this possible? The default install gives me this

> glxinfo | grep -i version
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
    Version: 19.2.1
    Max core profile version: 0.0
    Max compat profile version: 2.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.0
OpenGL version string: 2.1 Mesa 19.2.1
OpenGL shading language version string: 1.20
OpenGL ES profile version string: OpenGL ES 3.0 Mesa 19.2.1
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.00

An almost patched obs results in a core-dump - I'm missing the NEON translation for SSE's _MM_TRANSPOSE4_PS

> obs
info: Physical Cores: 4, Logical Cores: 4
info: Physical Memory: 3806MB Total, 2085MB Free
info: Kernel Version: Linux 4.19.80-v8-james
info: Distribution: "Ubuntu" "18.04"
info: Window System: X11.0, Vendor: The X.Org Foundation, Version: 1.19.6
info: Portable mode: false
QMetaObject::connectSlotsByName: No matching signal for on_advAudioProps_clicked()
QMetaObject::connectSlotsByName: No matching signal for on_advAudioProps_destroyed()
QMetaObject::connectSlotsByName: No matching signal for on_program_customContextMenuRequested(QPoint)
Attempted path: share/obs/obs-studio/themes/Dark/no_sources.svg
Attempted path: /usr/share/obs/obs-studio/themes/Dark/no_sources.svg
info: OBS 24.0.3-82-gb98ea73c-modified (linux)
info: ---------------------------------
info: ---------------------------------
info: audio settings reset:
        samples per sec: 44100
        speakers:        2
info: ---------------------------------
info: Initializing OpenGL...
[SegFault - Core dump]

TheRemote, 2019-11-02:
I think it's right on the edge of being possible but it's very bleeding edge.  This image has a special PPA that has us all the way up to mesa 19.2.1 (way ahead of everything else) because the v3d support for the Pi 4 is also so bleeding edge.  To see how bleeding edge, here's a good read from the Pi 4 v3d driver mesa guys: https://www.raspberrypi.org/blog/vc4-and-v3d-opengl-drivers-for-raspberry-pi-an-update/

Currently in the image if I go into kubuntu-desktop under the "Compositor" section of System Settings I can choose up to OpenGL 3.1 as my rendering backend for the compositor.  We are definitely hardware accelerated through 3.0 as glxinfo has confirmed.

The part I'm not sure about here is what does it take to get to 3.2?  From reading the article by the new dev it sounds like they're wanting to merge a bunch of 3.1 improvements in the next release.  I found another article at https://www.phoronix.com/scan.php?page=news_item&px=Raspberry-Pi-4-Almost-GLES-3.1 that says the same thing, they are about to release 3.1, so maybe it's not available yet?

Either way the hardware definitely *can* support 3.2.  From the Raspberry Pi blog article:

> The driver code for both GPU models is available in the Mesa upstream repository. The codename for the VideoCore IV driver is VC4, and the codename for the VideoCore VI driver is V3D. There are no downstream repositories – all development happens directly upstream, which has a number of benefits for end users:
> 
> It is relatively easy for the more adventurous users to experiment with development builds of the driver.
> It is fairly simple to follow development activities by tracking merge requests with the V3D and VC4 labels.
> At present, the V3D driver exposes OpenGL ES 3.0 and OpenGL 2.1. As I mentioned above, the VideoCore VI GPU can do OpenGL ES 3.2, but it can’t do OpenGL 3.0, so future feature work will focus on OpenGL ES.

So it sounds like *maybe* there's a development driver on the mesa site somewhere that could be built to get that far.

peardox, 2019-11-02:
Yeah, I've been reading those articles as well albeit after posting the query

I've got DEBs for Armv7a and aarch64 - both are incomplete while the _MM_TRANSPOSE4_PS macro is being written (sse2neon are doing that part)

As 19.10 is supposed to be bleeding-edge I've got a SD waiting to be try out the official Ubuntu version (at least I can find out what version of OpenGL it's got)

I'll try out the guys working on VC4 as OBS-Studio is something that'll probably be in high demand.

For now managing to pull together your stuff with a OpenGL 3.2 bleeding-edge, a desktop and OBS would make for a "interesting" image.

xlazom00, 2019-11-02:
If you want to try latest mesa you can try this ppa
It is mesa master 
https://launchpad.net/~oibaf/+archive/ubuntu/graphics-drivers

xlazom00, 2019-11-02:
I just update your image to 19.10
and then I installed ubuntu-desktop
And it is really usable!!!

TheRemote, 2019-11-02:
Great PPA, after installing I'm showing OpenGL ES 3.1 and OpenGL 2.1 renderers!

I'm switching my graphics driver PPA to this one since it's a couple versions newer.

peardox, 2019-11-02:
Anyone heard of Regal?

 https://github.com/p3/regal

xlazom00, 2019-11-02:
@peardox there is also gl4es

peardox, 2019-11-02:
Sadly that only does OpenGL 2.1

The version numbers for GL / GLES are misleading, you'd thing GLES 3.2 did GL 3.2...

TheRemote, 2019-11-04:
It looks like this has been answered, if anything changes or I missed something and there's more I can do let me know!

closed 2019-11-04 by TheRemote