TheRemote/Ubuntu-Server-raspi4-unofficial #69: Overclock

JeremyBYU, 2019-11-02:
Hi thanks for the image!
I was wondering how to overclock the CPU on this image. Before I modified config.txt under /boot on a raspian image. On this image there are two configs, one just named config and the other as config-4.19.80-etc . Which should I update?

Also uname -a gives
Linux ubu tu 4.19.80-v8-james #16. 
I would have guessed that the number should have been #17 for the release candidate I downloaded.Thanks again!

TheRemote, 2019-11-03:
Hey Jeremy,

On Ubuntu it's slightly different but you're crazy close.  In the boot folder there's another folder named "firmware".  So /boot/firmware/config.txt.  You'll recognize all the other folders in there!

closed 2019-11-04 by TheRemote

JeremyBYU, 2019-11-12:
Just commenting back that it worked. Thanks!