TheRemote/Ubuntu-Server-raspi4-unofficial #70: Package "<some_package>" has no installation candidate

CrackTheCode016, 2019-11-04:
Hi, first off thank you for this amazing image.

When attempting to install dependencies via `apt`, it seems `sources.list` does notcontain the usual links to said packages.  

Example:
```
Package libtool is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package autoconf is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package cmake is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'autoconf' has no installation candidate
E: Package 'libtool' has no installation candidate
E: Package 'cmake' has no installation candidate
E: Unable to locate package libatomic-ops-dev
E: Unable to locate package libunwind-dev
E: Unable to locate package libgflags-dev
E: Unable to locate package libsnappy-dev
E: Unable to locate package ninja-build
E: Unable to locate package python3-ply
E: Unable to locate package python-dev
E: Unable to locate package libxml2-dev
E: Unable to locate package libxslt-dev
```

What would be the best way to remedy this issue, besides tracking down each package's PPA link?

TheRemote, 2019-11-04:
Hey CrackTheCode,

That is definitely not normal and you won't have to track down any PPAs or anything like that.

Can you confirm that you did a sudo apt update first?  The only time I have seen messages like this is when I haven't done the sudo apt-get update first.  I do wipe the package cache in the image to reduce the download size so until that is ran there won't be anything in there.

If you've already done that could you paste a cat /etc/apt/sources.list?  I'm very curious what is in there!

CrackTheCode016, 2019-11-04:
Thanks for the fast response.  Yes, I've made sure to `sudo apt update`. 

Here is the output of  `sources.list`:

```
## Note, this file is written by cloud-init on first boot of an instance
## modifications made here will not survive a re-bundle.
## if you wish to make changes you can:
## a.) add 'apt_preserve_sources_list: true' to /etc/cloud/cloud.cfg
##     or do the same in user-data
## b.) add sources in /etc/apt/sources.list.d
## c.) make changes to template file /etc/cloud/templates/sources.list.tmpl

# See http://help.ubuntu.com/community/UpgradeNotes for how to upgrade to
# newer versions of the distribution.
deb http://ports.ubuntu.com/ubuntu-ports bionic main restricted
# deb-src http://ports.ubuntu.com/ubuntu-ports bionic main restricted

## Major bug fix updates produced after the final release of the
## distribution.
deb http://ports.ubuntu.com/ubuntu-ports bionic-updates main restricted
# deb-src http://ports.ubuntu.com/ubuntu-ports bionic-updates main restricted

## N.B. software from this repository is ENTIRELY UNSUPPORTED by the Ubuntu
## team. Also, please note that software in universe WILL NOT receive any
## review or updates from the Ubuntu security team.
deb http://ports.ubuntu.com/ubuntu-ports bionic universe
# deb-src http://ports.ubuntu.com/ubuntu-ports bionic universe
deb http://ports.ubuntu.com/ubuntu-ports bionic-updates universe
# deb-src http://ports.ubuntu.com/ubuntu-ports bionic-updates universe

## N.B. software from this repository is ENTIRELY UNSUPPORTED by the Ubuntu
## team, and may not be under a free licence. Please satisfy yourself as to
## your rights to use the software. Also, please note that software in
## multiverse WILL NOT receive any review or updates from the Ubuntu
## security team.
deb http://ports.ubuntu.com/ubuntu-ports bionic multiverse
# deb-src http://ports.ubuntu.com/ubuntu-ports bionic multiverse
deb http://ports.ubuntu.com/ubuntu-ports bionic-updates multiverse
# deb-src http://ports.ubuntu.com/ubuntu-ports bionic-updates multiverse

## N.B. software from this repository may not have been tested as
## extensively as that contained in the main release, although it includes
## newer versions of some applications which may provide useful features.
## Also, please note that software in backports WILL NOT receive any review
## or updates from the Ubuntu security team.
deb http://ports.ubuntu.com/ubuntu-ports bionic-backports main restricted universe multiverse
# deb-src http://ports.ubuntu.com/ubuntu-ports bionic-backports main restricted universe multiverse

## Uncomment the following two lines to add software from Canonical's
## 'partner' repository.
## This software is not part of Ubuntu, but is offered by Canonical and the
## respective vendors as a service to Ubuntu users.
# deb http://archive.canonical.com/ubuntu bionic partner
# deb-src http://archive.canonical.com/ubuntu bionic partner

deb http://ports.ubuntu.com/ubuntu-ports bionic-security main restricted
# deb-src http://ports.ubuntu.com/ubuntu-ports bionic-security main restricted
deb http://ports.ubuntu.com/ubuntu-ports bionic-security universe
# deb-src http://ports.ubuntu.com/ubuntu-ports bionic-security universe
deb http://ports.ubuntu.com/ubuntu-ports bionic-security multiverse
# deb-src http://ports.ubuntu.com/ubuntu-ports bionic-security multiverse
```

CrackTheCode016, 2019-11-04:
FYI, I am using the newest release (v18).

TheRemote, 2019-11-04:
Sources list looks good.  I'm firing up my Pi to see if I can reproduce on my end

TheRemote, 2019-11-04:
So far I have not been able to get it to do this.  Can you paste the output of your apt-get update?  

Let's also try rebuilding your /etc/apt/sources.list file:

CrackTheCode016, 2019-11-04:
It appears rebuilding my `source.list` did the trick!!  Before, my `apt-get update` only showed a few Hits, and no Gets.  Now it updated normally!

The only config I did was disable `cloud-init`.  Could that have affected it?

TheRemote, 2019-11-04:
It's possible, I know that whole file is created by cloud-init but I thought (and the comments at the top of the file say) that it is generated upon first boot and then left alone.

How did you disable it?  I'd like to reproduce it if I can

CrackTheCode016, 2019-11-04:
I disabled as per `cloud-init` docs - by placing an empty file called `cloud-init.disabled` in `/etc/cloud/`:

`sudo touch /etc/cloud/cloud-init.disabled`

TheRemote, 2019-11-04:
Perfect, thanks!

I'll add a fix for this in the next release.

closed 2019-11-04 by TheRemote