TheRemote/Ubuntu-Server-raspi4-unofficial #72: ubuntu-18.04.3-preinstalled-server-arm64+raspi4.img Failed 3 lines after "-END of Host Key Keys-"

tikiatsbme, 2019-11-04:
Hi, I tried again with V17 as ubuntu-18.04.3-preinstalled-server-arm64+raspi4.img.  I Did:

sufo apt-get install git
git clone https://github.com/TheRemote/Ubuntu-Server-raspi4-unofficial
download V18 - current 20191104
archiver to expand image
run gparted and format micro sd to fat 32
sudo dd bs=4M if=file.img of=/dev/sdd status=progress conv=fsynce
copy some files to HOME file on sdd
copy directory
sudo rsync -av /someplaceold/piclone /someplacenew/piclone
Shutdown, switch cards, start Rpi4B_4GB

#### ERROR ######
Failed to continue  3 lines after "-END of Host Key Keys-"
2nd line said:  ci-info, no authorized ssh keys fingerprints found for user ubuntu

Log file to be attached (Edit pending)

TheRemote, 2019-11-08:
Sorry I have been slow to answer this, I have been busy.  I'd like to suggest an alternate approach of mounting the image to do this:

```
sudo apt install kpartx
sudo kpartx -avs "file.img"
MountCheck=$(sudo kpartx -avs "file.img")
echo "$MountCheck"
export MOUNT_IMG=$(echo "$MountCheck" | awk 'NR==1{ print $3 }')
export MOUNT_IMG="${MOUNT_IMG%p1}"
sudo mount "/dev/mapper/${MOUNT_IMG}p2" /mnt
sudo mount "/dev/mapper/${MOUNT_IMG}p1" /mnt/boot/firmware
```

The image is now mounted.  If you go to cd /mnt it will be the whole image.  Now copy piclone over:

```
sudo rsync -av /someplaceold/piclone /mnt/usr/bin/piclone

```

When you are finished copying everything you want to on the image close it like this:

```
sudo umount /mnt/boot/firmware
sudo umount /mnt
sudo kpartx -dvs "file.img"
```

Now you can burn the file.img normally with dd.  Try this way and see if the boot errors go away!

tikiatsbme, 2019-11-08:
I will try a CLEAN git clone and a clean DD, let it boot to curser. I have been altering image, ooops, sorry, learning fast though.

closed 2019-11-08 by tikiatsbme