TheRemote/Ubuntu-Server-raspi4-unofficial #74: machine-id is not unique!

cannfoddr, 2019-11-05:
I am trying to use your image for a 5 node pi 4 k8s cluster.  I have been struggling to get networking working with weave net cni plugin.  Looking at the logs I was seeing errors about duplicate identifiers.

Digging a bit deeper I was directed to check the machine identifiers in /etc/machine-id - at this point I discovered that every Pi had exactly the same machine-id:

> 1c4ad377574c4263adbb7c624b6a3429

I suspect this has been cloned rather than generated at startup of the image.  

I am off to figure out how to get these to be unique and start the cluster build again from scratch with unique ids

TheRemote, 2019-11-06:
Hey cannfoddr,

I would think that cloud-init's first startup would have taken care of this.  I can definitely remove it during the image creation process so it is forced to make a new one.

At any rate, this is extremely easy to deal with.  Just issue a:

sudo rm /etc/machine-id
sudo systemd-machine-id-setup

It will tell you it has generated a new system ID and if you cat /etc/machine-id it will have changed.

cannfoddr, 2019-11-06:
I tried deleting the file and rebooting which turned out to be bad!  

I settled for:

`sudo echo -n > /etc/machine-id`

followed by a reboot which seemed to fix it.

Problem manifested when weave net in k8s reported duplicate ids and the weave network adapters it created had the same physical ids on all 5 nodes.  Container creation for new pods was stuck waiting for network from CNI I think.

As far as I recall they all went though cloud-init at initial startup.

TheRemote, 2019-11-07:
Ahhh, good point, I also did a sudo touch /etc/machine-id in the script since I saw that the original image had the file but it was 0 bytes!

I put this fix in V19 to prevent this from happening to anyone else, thanks for reporting it!

closed 2019-11-07 by TheRemote