TheRemote/Ubuntu-Server-raspi4-unofficial #75: Removing old kernel source

freewill404, 2019-11-06:
Getting this after running ./Updater.sh:

`Note: it is recommended to periodically clean out the old kernel source from /usr/src, it's quite large!`

Do I just rm all the directories under /usr/src, including the latest one "4.19.81-v8-19", or the newer one I don't touch?

TheRemote, 2019-11-07:
Good question, that message is basically temporary as I'm working on using .deb packages to install this stuff so they will be much easier to remove.

The best thing to do is remove all the old ones.  If you know you aren't building any kernel or DKMS modules you can remove the newest one too if you want to save some space.

sudo rm -rf 4.19.81-v8-18 

Would remove the v18 source for example and save about 800MB of space!

TheRemote, 2019-11-08:
This is included in v21 by default, closing

closed 2019-11-08 by TheRemote