TheRemote/Ubuntu-Server-raspi4-unofficial #76: Unable to locate package libraspberrypi-bin

peardox, 2019-11-07:
Sounds like a problem...
See below - just above the --- break ---
====================== Log ======================
HEAD is now at 0b8abfe v20 Release
Updater is up to date.  Checking system ...
Release v20 is available!  Make sure you have made a full backup.
Note: your /boot cmdline.txt and config.txt files will be reset to the newest version.  Make a backup of those first!
Update now? (y/n)y
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package libraspberrypi-bin
Downloading update package ...
-------------------------------------------- break --------------------------------------------
[lots of activity - not in buffer after it finished]
-------------------------------------------- break --------------------------------------------
Updating kernel and modules ...
update-initramfs: Generating /boot/initrd.img-4.19.81-v8-20
Cleaning up downloaded files ...
rm: cannot remove '/etc/ubuntufixes.sh': No such file or directory
rm: cannot remove '/etc/init.d/ubuntufixes': No such file or directory
Running Ubuntu fixes ...
Attaching Bluetooth controller ...
bcm43xx_init
Set Controller UART speed to 921600 bit/s
Flash firmware /etc/firmware/brcm/BCM4345C0.hcd
Set Controller UART speed to 921600 bit/s
Device setup complete
Ubuntu fixes complete ...
Removing old Ubuntu fix file ...
Update completed!
Note: it is recommended to periodically clean out the old kernel source from /usr/src, it's quite large!
You should now reboot the system.

TheRemote, 2019-11-07:
Hey peardox,

Sorry for the lack of clarity here, I'll explain then I'll make those messages not show up in the Updater output.

In old versions of the image we included a very outdated version of raspi-config and libraspberrypi-bin (which is things like vcgencmd, etc).  The image has been updated and includes the very latest of these utilities which are built fresh from the latest source in BuildPiKernel64bit.sh.

So basically the updater is looking for the old outdated versions of these utilities that were previously included and removing them but the output is very misleading!

TheRemote, 2019-11-08:
Fixed in V21

closed 2019-11-08 by TheRemote