TheRemote/Ubuntu-Server-raspi4-unofficial #78: V20 failed to complete boot three times

tikiatsbme, 2019-11-08:
I did this:
inserted sd card into reader
Gparted, unmounted two partitions, deleted two partitions, created 1 new fat 32 partition, applied. Success.
Gparted, checked new partition, no errors.
******
Removed old JCDirc to trash (CLEAN)
Git Clone https;//JCweblin.git  - success
clicked and downloaded JCimg.img.xz
Opened archiver, and extracted IMG file (369.4M to 3.2GB)
sudo dd bs=4M if=JCimage.img of=/dev/sdd staus=progress conv=fsync
wrote 3,431,056,384 bytes in 21.383 sec at 16.2 MB/s
ran df -h (not mounted)
removed sd card
shutdown PI4B_4GB, removed micro sd
inserted JCimg sd
tried USB plugged in, failed at 3 lines after SSH Key KEYS echo.
power off, power on
Tried no USB except kybd and mouse, failed 1 line after login
power off, power on
tried with all USB attached again (this is your syslog below)
FAILED 1 line after login

AFTER FAIL, removed card, inserted Raspbian 32Bit, booted to desktop
Inserted JCIMage sd card to view partitions
System boot has 213 files with 41998067 bytes
Writeable has 115100 files with 2115265068 bytes

OK, what am I doing wrong? Is sd card bad?
I will try sd card copy and see if raspbian 32 will boot from it.

[20191108V20_3dreboot_syslog.log](https://github.com/TheRemote/Ubuntu-Server-raspi4-unofficial/files/3826622/20191108V20_3dreboot_syslog.log)

TheRemote, 2019-11-09:
Is this a modified image or the original?

TheRemote, 2019-11-09:
The partitions in Ubuntu are mounted by label, the ext4 partititons label should be "writable" and the fat32 label will be "system-boot".  I see you are recreating blank partitions yourself which won't have these labels.

I'd try writing the image to the SD card using Etcher from https://www.balena.io/etcher/

tikiatsbme, 2019-11-09:
Answer to above: They are dd block by block copies of YOUR orginal.
Answer: ETC4 is indeed Writable while FAT32 is System-Boot, Correct x 2
######################
Attached file is testing input/output in terminal of subject JCIMG.IMG sd card

SUUMARY: no issues, SD micro card is AOK! so what is going on, is it some weird issue with early RPi4B's with 4 GB?

What I did:

I ran SD Card Copy and then inserted sd micro into RPi4B, it booted to desktop - OK
I then switched back to my regular sd card and inserted the questionable card into reader.
I umounted SDD1, SDD7.  Note that sdd5 and sdd6 were not mounted 
I ran fsck on sdd and no issues
I ran badblocks for 2 hrs, no issues (0/0/0)

[terminal_Output_Sdd.txt](https://github.com/TheRemote/Ubuntu-Server-raspi4-unofficial/files/3826863/terminal_Output_Sdd.txt)

tikiatsbme, 2019-11-09:
I successly (Congrats to James!) installed ubuntu-18.04.3-preinstalled-desktop-arm64+raspi4.img and it WORKED the first time, so closing this one as I am not into servers.

On a side note: I donot like SNAPS in ubuntu, and ther are no builds for RCLONE GUI so its still an experiment. Yet, a BIG THANKS, the desktop img works!!!!!

closed 2019-11-09 by tikiatsbme