TheRemote/Ubuntu-Server-raspi4-unofficial #79: USRU Wallpaper?

peardox, 2019-11-09:
Just a thought.

The 'proper' releases have a desktop background for their new versions. Why not have one for yours?

Be fun playing spot the logo on youTube videos :)

TheRemote, 2019-11-11:
I love the idea!

At first I was trying to keep the release as close to official as possible but at this point we have so many fixes that official doesn't have (18.04 and even 19.10) that we continue to deviate and become something else every release.

A lot of the problems we have fixed in this branch are bugs that have been in official *forever* and probably still will be for years to come.  Maybe it's time to embrace that fact!

peardox, 2019-11-11:
I thought so too

Sounds like you just need to add image(s) to /usr/share/backgrounds

[Without bothering to look it up] Wonder if the default's easy to set?

If you did a full desktop release I guess it'd get listed somewhere :) The Ubuntu Mate guy is taking ages to do his version I notice and the official Ubu needs extensive patching...

closed 2022-05-11 by TheRemote