TheRemote/Ubuntu-Server-raspi4-unofficial #80: Excessively high 'polkitd' CPU usage under ubuntu-desktop

gimmx, 2019-11-10:
Followed your instructions on a Raspberry Pi 4B running xorg with ubuntu-desktop and 'polkitd' runs at 60-90% CPU, making the connection almost worthless.

TheRemote, 2019-11-11:
Interesting.  I think this is a groups issue (either regular groups or possibly udev groups) as I have seen warnings in the logs about xrdp groups possibly missing/configured incorrectly.  I'll put this on the fix list for the next release and see if we can't get that service fixed up.

TheRemote, 2019-11-11:
So I haven't been able to reproduce this yet.  Let me outline my steps.

Wrote my preinstalled desktop v21 image to SD card
Started up Pi (connected to a monitor)
SSH'd in with Putty and installed xrdp using apt install xrdp
Connected with RDP from Windows 10, signed in with default Xorg option, ubuntu/newpass entered as account

It went right in, CPU usage at about 2-8%.  Screenshot attached.  What did I do differently?
[image]

TheRemote, 2019-11-11:
Ok, I tried this headless with no monitor attached and am getting the high polkitd usage, but no authentication issues, so it appears at least the polkitd issue is triggered by not having any monitor attached to the Pi

gimmx, 2019-11-11:
Yes I should've been more specific. I'm running headless.

TheRemote, 2019-11-11:
No worries, I am now seeing it too, if I connect it to a monitor and reboot everything goes back to normal so it appears that it is not happy at all if it's headless.  I'll see what we can do to get that fixed!

TheRemote, 2019-11-17:
So this one is turning out to  be a little bit of a can of worms.  The problem seems to be that without a display connected the video subsystem isn't loading which gets into drivers and things like that.

I'm going to leave this open until I nail it but it may take some more time as it goes pretty deep!

TheRemote, 2019-11-22:
I've made some progress here, can you try with v23?

TheRemote, 2019-11-22:
This has been fixed in v23

closed 2019-11-22 by TheRemote