TheRemote/Ubuntu-Server-raspi4-unofficial #82: Problem with lightdm-greeter-session for v20 build

zebrastack, 2019-11-10:
I have been using xubuntu-desktop since long time with your server type images. Today i tried to install xubuntu-desktop over v21 image and i noticed that lightdm-greeter-session is hanging on loading logon screen it only shows mouse but nothing more no screen. It was working on v19 image.

zebrastack, 2019-11-10:
I notice a error on /var/log/lightdm/seat0-greeter.log like WARNING: [PIDs] Failed to execute command: upstart

I only setup netplan to access internet with raspberry pi and "apt update && apt install -y xubuntu-desktop" commands over server image

TheRemote, 2019-11-11:
Hey zebrastack,

Thanks for the report, indeed this was working just a release or two ago so I will install it and see what changed and get a fix out!

TheRemote, 2019-11-11:
I can't explain why it's happening now and didn't before, but I have a fix for you:

 sudo nano /etc/X11/xorg.conf

The file should not exist.  Insert the following:

Section "Device"
        Identifier  "Card0"
        Driver      "fbdev"
EndSection

Then Ctrl+X and save yes

now:

sudo systemctl restart lightdm

It will pop right up as normal.  This fix will be in v22.

zebrastack, 2019-11-11:
Hi, issue is resolved after following your recommendation. Thanks for all your effort & support.

closed 2019-11-11 by zebrastack