TheRemote/Ubuntu-Server-raspi4-unofficial #83: Ubuntu Core

wyuenho, 2019-11-11:
What's the process of building [Ubuntu Core](https://ubuntu.com/download/iot/raspberry-pi-2-3-core) for the Pi 4? Is it possible to release prebuilt versions of it based off this current process?

TheRemote, 2019-11-11:
I think the current process would barely be affected by Ubuntu Core.  The apt-get upgrade parts wouldn't work on it (since Core is completely snap based) but the most important parts of getting it running like copying the updated firmware, etc. I don't think would change.

Why would you want Ubuntu Core over a barebones Ubuntu Server (which has apt, etc)?  Don't take this question the wrong way such as me trying to say I know better or anything like that, I am genuinely curious and am not as familiar with Ubuntu Core and what the use cases are/what people use it for.  If there is demand and good use cases to choose Ubuntu Core over anything else I can definitely look into providing a binary for it!

wyuenho, 2019-11-14:
> Why would you want Ubuntu Core over a barebones Ubuntu Server (which has apt, etc)? 

I heard it's only about 100m larger than Alpine, and it's preconfigured to on resource constrained devices like the Pi (i.e. hopefully no useless packages like btrfsprogs), so I wanted to try it out, but there doesn't seem to be a pre-made image for the Pi 4 out there, so I thought I'd ask here.

TheRemote, 2019-11-22:
We'd have to cut down the firmware quite a bit and a few other things for it to be small, so it's probably beyond the scope of this project, but I'm open to it if more requests come in

closed 2019-11-22 by TheRemote