TheRemote/Ubuntu-Server-raspi4-unofficial #84: rtl8812au support?

knro, 2019-11-11:
This is not a bug but a request, can this driver included with the build?

https://github.com/gnab/rtl8812au

Are there instructions for build it for arm64 or just the usual instructions indicated above?

TheRemote, 2019-11-11:
Hey knro,

The best way to build this would be the dkms module.  I include the entire source tree in /usr/src to make building any modules like this both possible and easy to do.

You would skip all the top parts of the README and go straight to the DKMS section.  You'd copy the source to /usr/src/8812au-4.2.2 and then 

```
sudo dkms add -m 8812au -v 4.2.2
sudo dkms build -m 8812au -v 4.2.2
sudo dkms install -m 8812au -v 4.2.2
```

The nice thing about dkms is it will automatically rebuild with any updates and is generally pretty easy to use.  Definitely give the DKMS method a try and let me know!

knro, 2019-11-11:
Thanks for the quick answer! I just tried this but it fails:

4.19.81-v8-20  8812au-4.2.2
ubuntu@ubuntu:/usr/src$ cd 8812au-4.2.2/
ubuntu@ubuntu:/usr/src/8812au-4.2.2$ sudo dkms build  -m 8812au -v 4.2.2

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' all KVER=4.19.81-v8-20...(bad exit status: 2)
ERROR (dkms apport): binary package for 8812au: 4.2.2 not found
Error! Bad return status for module build on kernel: 4.19.81-v8-20 (aarch64)
Consult /var/lib/dkms/8812au/4.2.2/build/make.log for more information.
ubuntu@ubuntu:/usr/src/8812au-4.2.2$ cat /var/lib/dkms/8812au/4.2.2/build/make.log
DKMS make.log for 8812au-4.2.2 for kernel 4.19.81-v8-20 (aarch64)
Mon Nov 11 22:44:33 +03 2019
make ARCH=aarch64 CROSS_COMPILE= -C /lib/modules/4.19.81-v8-20/build M=/var/lib/dkms/8812au/4.2.2/build  modules
make[1]: Entering directory '/usr/src/4.19.81-v8-20'
Makefile:614: arch/aarch64/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/aarch64/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/4.19.81-v8-20'
Makefile:1065: recipe for target 'modules' failed
make: *** [modules] Error 2
ubuntu@ubuntu:/usr/src/8812au-4.2.2$ ls
Kconfig  Makefile  README.md  clean  core  dkms.conf  hal  ifcfg-wlan0  include  install.sh  os_dep  project  repoTest1.js  repoTest2.htm  runwpa  wlan0dhcp

knro, 2019-11-12:
I fixed it by changing architecture to arm64, thanks!

closed 2019-11-12 by knro