TheRemote/Ubuntu-Server-raspi4-unofficial #85: Stock VNC 10.04 screen mirroring bug

peardox, 2019-11-11:
There's been a problem in UBU 18.04 since it's release and it really affects how I + presumably others use my / our Pi(s)

Basically getting VNC to run at all is a major pain. Raspbian on the other hand has a raspi-config option to turn on VNC that just works - licensed from RealVNC - that mirrors your desktop you a VNC client.

My Pi(s) connect to my TV, which is about 12 feet away making it hard to read (I'm getting old - eyesight not what it was)

The standard UBU method uses Vino and is supposed to do the same thing as Pi do but it's all screwed up - packages not installed, gnu keys not set and other issues. The desktop settings icon for Sharing crashes, badly - that's the first sign of the troubles that await...

The perfect solution would be fixing the pre-installed system, a work-around would be to do it with another VNC server.

NVidia use 18.04 for the Nano - the functionality is broken there as well (you'd have thought NVidia would want the Nano to work properly - I guess not)

TheRemote, 2019-11-17:
Hey peardox,

I agree that the Raspbian RealVNC system is beautiful.  They're basically giving the $40/device/year license out for free!

So when you talk about fixing the built-in system is that Vino or which packages are involved?  When I was playing with xrdp I saw some people using tightvncserver, then there's also a couple of other ones.

I'd like to start playing with this and see if I can get the built in system working or explore alternatives if it makes sense to do so!

peardox, 2019-11-17:
Yeah, it's Vino and they screwed it all up in 18.04 - I believe it was just as seamless as Raspbian in 16.04

I've got an x86 box running 16.04 and another on 18.04 - I'll see what I can discover. There are a few articles that are supposed to explain how to fix things - none of them worked for me. This is, as I say, the preferred solution as it's hooked into the Ubu way of doing things anyway.

An alternative suggested in the Pi forums was using another VNC and doing the same with that - got fairly far with that method and did manage mirroring but not on start-up (something I screwed up I guess in that try)

I'll do some research and get back to you with any results know I know you're interested.

TheRemote, 2019-11-22:
Hey peardox,

Can you try v23 just out of curiosity?  I made some changes so that there would be a default display loaded even headless.  I'd like to see if it is any less broken.  If not I will dive into this for v24.

peardox, 2019-11-22:
Cool

No wallpaper yet :(

Got Unity installed and Screen Sharing comes up - you just can't turn it on

Googled this - https://askubuntu.com/questions/1060457/ubuntu-18-04-1-lts-cant-enable-screen-sharing

Follow the simple instructions re managed network + netplan and after a reboot I can turn it on and VNC almost connects (wish my Nano got this far)

I'm using RealVNC viewer as that's what RPi use and it's complaining about the encryption being too strong.

Gonna look at putting Vino into ultra-insecure mode + also try a diffo VNC client

peardox, 2019-11-22:
Making VNC ultra-insecure

Dead simple (google it - answer on first page I checked)

gsettings set org.gnome.Vino require-encryption false

Anyway, this gives the desired result and as it's on a local net with no incoming outside connectivity and is... well... a Pi it's good enough for me

I'll also try it with encryption and a diffo client

[image]

peardox, 2019-11-22:
Done some real world user (ME) testing

Installed kubuntu + xubuntu - eek - diffo VNC server so went back to ubuntu and nothing worked (damn)

Oh well, wanted to try a fresh install anyway... Ooh - a desktop version, lets give that a go...

Install to SD - now on a completely virginal build

Added my little mods as per above + everything works perfect - I can even record Pi action in Windows with OBS etc.

Added a simon user and set myself sudo, have a look around - ooh, a james user in home

Reboot...

OK - VNC server not running at login screen so have to switch TV to Pi4 to login so VNC starts... bummer. One logged in VNC kicks in and everything hunky dory

So...

All very cool + swishy

Desired mods...
1) Jubuntu (James' Ubuntu) Wallpaper!!!
2) Enable VNC on login screen
3) Apply the mods above

Q: [Thinking out loud] Would it be possible to have a file in boot have VNC as enabled by default? Raspian now enables you to add a file called ssh to boot which enables ssh server on startup for headless. Headless extrapolation would be enabling VNC optionally (not thought this thru fully)

TheRemote, 2019-11-22:
Very cool testing!  I definitely want to integrate these and improve the VNC experience.  V23 fixed up the xRDP experience quite a bit.

I am not sure where that james folder is coming from exactly.  I must have a typo in one of my sudo commands or maybe in a chroot that is pulling the name from my username on my test VM!  I'm guessing if someone else ran the build script they would also end up with a folder with the username they use on their build machine as well.  I'll see if I can track that down!

peardox, 2019-11-22:
There's a home dir but nothing in passwd, shadow, group related to /home/james

The (very little) fixes I did today make JUbuntu cool from a.n.other machine for users who expect a GUI (which is most of them)

peardox, 2019-11-22:
Oh yeah - one of the more popular Pi youtubers did Ubu 19.04 the other day

https://www.youtube.com/watch?v=cPk1pvgK7JA

I wonder if he knows how broken / soon to expire it is?

TheRemote, 2019-11-22:
Yeah that release doesn't make any sense to use!  If you want long term support you should be on LTS.

If you're going to go outside LTS you might as well jump all the way to Focal Nightlies at http://cdimage.ubuntu.com/ubuntu-server/daily-preinstalled/current/HEADER.html since we know that the kernel fixes for 19.10 are still getting checked/approved and they should hit there first if they haven't already.

bruces22, 2019-11-24:
Krfb on kde seems to be a good vnc app.  Easy setup, with vnc client.

peardox, 2019-11-24:
Yeah, tried that when doing multiple desktop tests

The point is making VNC just happen like the Pi does so the built-in (Vino) bit in Settings makes sense. It's actially probably slightly more noob friendly than Pi which instructs you to use raspi-config ... which  is confusing anyway - bad menu names, never can find what I'm after in the damned thing

closed 2022-05-11 by TheRemote