TheRemote/Ubuntu-Server-raspi4-unofficial #86: How to connect to Wi-Fi by config file in boot partiton?

VoidXia, 2019-11-13:
I want to connect to my pi wirelessly without ever having a network cable or a monitor. In raspbian I can directly create wpa_supplicant.conf in the boot partition on SD card to make pi connect to my Wi-Fi once launched. I wonder if that is possible in this modified ubuntu firmware?

wyuenho, 2019-11-17:
You just edit the [netplan](https://netplan.io/reference) file in /etc/netplan

TheRemote, 2019-11-17:
Yes, netplan needs to be used in Ubuntu.  I thought about removing netplan at one point but part of it being Ubuntu is that it has netplan which has upsides and downsides.

I am going to leave this one alone because we have tweaked netplan in the past and it really messes with a lot of use cases people have reported.

closed 2019-11-17 by TheRemote