TheRemote/Ubuntu-Server-raspi4-unofficial #87: i2c issues

fjp, 2019-11-17:
Hi, thank you for the arm64 version. I am using your pre installed desktop [release v21](https://github.com/TheRemote/Ubuntu-Server-raspi4-unofficial/releases/tag/v21).
Everything is working so far but I couldn't get i2c working. 

I try to connect a grove oled display but when I execute `sudo i2cdetect -y 1` I get the following error:

> Error: Could not open file `/dev/i2c/1' or `/dev/i2c/1': No such file or directory

Using `sudo raspi-config` and enabeling i2c I got rid of this error. However, I still cannot see that the oled display is connected.

fjp, 2019-11-17:
Ok I found the issue. I connected the device to i2c 1 (physical pins 27 and 28, BCM 0 and 1, see [pinout.xyz](https://pinout.xyz/pinout/i2c)). After I connected the oled display to i2c 0 `i2cdetect` finds the device and I can send text to the display.

Is it normal that i2c 0 is not working? How is it possible to enable both i2c pins?

TheRemote, 2019-11-17:
Hey fjp,

If I'm reading this right then i2c 0 works but only if you manually run the i2cdetect command on it?

It may be a udev rule that isn't present in Ubuntu but is in Raspbian

fjp, 2019-11-17:
Hi @TheRemote sorry for the confusion, it was rather late when I tried this. I slightly edited my previous post. i2c0 (physical pins 3 and 5) seemed to be not working but I will try again soon to verify that both i2c interfaces are working.

fjp, 2019-11-18:
@TheRemote I can confirm that I have troubles to get i2c0 ([physical pins 27 and 28](https://pinout.xyz/pinout/i2c#)) working. Here is the output of `i2cdetect` for both i2c interfaces:

``` 
fjp@ubuntu:~$ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- 3c -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
fjp@ubuntu:~$ sudo i2cdetect -y 0
Error: Could not open file `/dev/i2c-0' or `/dev/i2c/0': No such file or directory
```

TheRemote, 2019-11-22:
This can be fixed by adding the following to config.txt:

dtparam=i2c1=on
dtparam=i2c0=on

TheRemote, 2019-11-22:
I added these options into the config.txt to make them easy to find.

Marking as fixed for v23.

closed 2019-11-22 by TheRemote