TheRemote/Ubuntu-Server-raspi4-unofficial #88: /proc/cpuinfo isn't correct

mpoly, 2019-11-18:
I noticed /proc/cpuinfo returns
Hardware	: BCM2835
instead of BCM2711
And in processors list, "model name" is missing.
I have some doubts about "Features" too, as Neon isn't in the list.
Regards.

czirok, 2019-11-19:
> Note: As of the 4.9 kernel, all Pis report BCM2835, even those with BCM2836, BCM2837 and BCM2711 processors. You should not use this string to detect the processor.

[https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md](url)

kaiomatico, 2019-11-21:
I think neon is not in the list because armv8 always includes a fpu automaticially by specification. So there actually is no neon feature listed, because in armv8 it is already baked in and mandatory.

closed 2019-11-22 by TheRemote