TheRemote/Ubuntu-Server-raspi4-unofficial #89: apparmor module is not loaded

deltasierra-ufc, 2019-11-21:
Using v22 release, fairly fresh install.

This is what I get when I submit 

> sudo apparmor_status

> apparmor module is not loaded

I've gone thru the Ubuntu apparmor wiki, starting the service, making sure I have apparmor, apparmor_utils, and apparmor_profiles installed.  Running 

> sudo modprobe appmor

 results in 

> FATAL: Module apparmor not found in directory /lib/modules/4.19.83-v8-22

How can I enable the apparmor module?  I also had issues trying to run Firefox and other apps with firejail installed -- could this be the reason why?

TheRemote, 2019-11-22:
Possibly, I have experimented with enabling AppArmor in the kernel before, I will enable it for the next build then let's check

TheRemote, 2019-11-22:
Fixed in v23

closed 2019-11-22 by TheRemote

deltasierra-ufc, 2019-11-22:
Ok, awesome!  Thank you James.