TheRemote/Ubuntu-Server-raspi4-unofficial #90: OpenGL issue

tarsij, 2019-11-24:
Hi TheRemote,

Since v22 the OpenGL driver used is wrong.

When I check the 'glxgears -info' for v21 it reports:
GL_RENDERER   = V3D 4.2
GL_VERSION    = 2.1 Mesa 20.0.0-devel (git-2e14fe6 2019-11-22 bionic-oibaf-ppa)
GL_VENDOR     = Broadcom

When I do the same in v23 I get:
GL_RENDERER   = llvmpipe (LLVM 9.0.0, 128 bits)
GL_VERSION    = 3.1 Mesa 20.0.0-devel (git-a2689eb 2019-11-23 bionic-oibaf-ppa)
GL_VENDOR     = VMware, Inc.

Some symphtoms I've observed:
 - WebGL aquarium test produces less FPS
 - SMPlayer instead of playing the movie in the main window it opens a new mpv one without controls
 - movie is not smooth in full screen
 - high processor usage is shown in the system monitor when watching movies
 - youtube is unusable

BTW thanks for the effort of putting this all together!

TheRemote, 2019-11-29:
Hey tarsij,

I'm not completely sure.  I just tried on my development build of v24 that I'm preparing and I have V3D 4.2 and 2.1 mesa again.

Let's try yours on v24 when I release it shortly here!

tarsij, 2019-11-30:
Hey TheRemote,

Sure, let's give it a try.

For the record I add some extra info (I saw some other guys had issues with v22/v23 as well):
- before starting ot use your images I refreshed the eeprom (since then I haven't done it, so I might give it a try to see if that solves the issue)
- I started to use your images from v18. I burnt the images with Balena Etcher.
- I use ubuntu-mate-desktop
- whenever you released, I updated to it with the Updater.sh.
- when I upgraded to v22 I started to experience some issues with the movies. I tried to fiddle without success with the opengl drivers and downgraded them to an older version as I thought that's the rootcause. Then I tried to upgrade to v23 hoping that it'll fix the issue, but the situation was the same. After this I decided to start from a fresh image as I couldn't figure out what the problem was. I burnt the v23 to an SD, then on the first run I did an upgrade + ubuntu-mate. I haven't installed anything else just checked the glxinfo and aquarium webgl test. It was the wrong one/slow. Then I went back to v22 and did the very same things with the very same results. Then I tried the v21 which was all good. Since then I use this image (+added an SSD as per your desription) and it is working fine.

Cheers

tarsij, 2019-11-30:
Hey TheRemote,

I've updated the system to the v24 and the situation is still the same. glxgears reports:
GL_RENDERER = llvmpipe (LLVM 9.0.0, 128 bits)
GL_VERSION = 3.1 Mesa 20.0.0-devel (git-a2689eb 2019-11-23 bionic-oibaf-ppa)
GL_VENDOR = VMware, Inc.
Aquarium WebGL is at 4-6 FPS instead of 15-17 and the processor is burning.

I can try to update the eeprom, what do you think can that help?
Any idea what else can I try?

Regard

tarsij, 2019-11-30:
Some follow up,

I've updated the eeprom to the latest (18 Nov) and just in case rerun the Updater.sh
Still the same...

Do you want me to check anything before I revert to v21?

Regards

tarsij, 2019-12-02:
Hey TheRemote,

Looks like I've managed to sort out the issue.

After the upgrade to the v22 xorg has created the /etc/X11/xorg.conf file with the following content:
Section "Device"
Identifier "Card0"
Driver "fbdev"
EndSection

I had to change the ' Driver "fbdev" ' line to ' Driver "modesetting" '

Now everything looks ok.

Cheers

closed 2019-12-02 by tarsij

TheRemote, 2019-12-02:
Hey tarsij,

Great find!  I'll add your fix into V25!

deltasierra-ufc, 2019-12-02:
tarsij,

   I did have this issue as well!  😲 I updated that text file as you mentioned and the issue was corrected for me. 😃 

   Thanks much!  Great work in finding that out.

James:  glad to hear it'll be included in V25.

tarsij, 2019-12-03:
I'm glad I could contribute to the project :D

tarsij, 2019-12-04:
Hey TheRemote,

Thanks for the new release!
Related the fix... If I remember right the /etc/X11/xorg.conf file is recreated whenever it's missing, so deleting it won't fix the issue for those who haven't changed that entry.

Regards