TheRemote/Ubuntu-Server-raspi4-unofficial #92: desktop not loading - upgradin from v22 to v23

giulioturrisi, 2019-11-25:
I just updated my raspberry trough the Updater.sh to V23. After reboot, i can login, use the terminal and so on, but i cannot see any folder - see any apps and so on!

giulioturrisi, 2019-11-25:
I tried to install from scratch the v23, and i got the same behavior!

Cherkes001, 2019-11-26:
I confirm that I installed version 23 from scratch and it does not boot, it’s just a black screen, even the bootloader did not try to start.

deltasierra-ufc, 2019-11-26:
I'm not having any issues with the Ubuntu MATE desktop.  _Just for feedback purposes_.

Cherkes001, 2019-11-26:
I installed the server version, then decided to check it again and installed version 22, it loaded, but the error `timeout waiting for hardware interrupt error` infinitely infuses into the terminal.
Could this mean that my flash drive is too slow?

TheRemote, 2019-11-30:
I'm not having any issues either.  A very small group of people have been reporting issues like this for months and I can't figure out what they're doing wrong or what is different.

When I build the image I write it to a SD card with Win32DiskImager or Etcher and plug it into my Raspberry Pi 4 to test it.  All Raspberry Pis should be largely the same.

All I can think is that maybe they have a really old bootloader and have never ran the Raspbian bootloader firmware updater?  Make sure you have updated the firmware to the latest using Raspbian (see my guide here: https://jamesachambers.com/raspberry-pi-4-bootloader-firmware-updating-recovery-guide/)

For Cherkes001, what are you doing to prepare it to boot on a flash drive?  Do you have a Raspberry Pi 4?  Because they don't boot from USB yet unless you use a SD card as a bootloader (see my article at https://jamesachambers.com/raspberry-pi-4-usb-boot-config-guide-for-ssd-flash-drives/).

Cherkes001, 2019-11-30:
Yes, I have a Raspberry Pi 4 (4GB), I record the image through Etcher. Perhaps my problem is in a bad SD card, and I updated the bootloader two or three times from the time of purchase. Well, maybe it's worth checking the boot from USB, because I have an external hard drive with USB 3.0, I think it will work better all the same.

tarsij, 2019-12-01:
> I'm not having any issues with the Ubuntu MATE desktop. _Just for feedback purposes_.

I'm using Ubuntu Mate as well and the desktop is loading although since v22 I have issues with the opengl driver, it is not recognised/used. glxgears -info reports a llvmpipe instead of V3D 4.2. I'm wondering if it's an overlay issue.

for deltasierra-ufc, may I ask you whether you experience similar issues with the opengl driver?

giulioturrisi, 2019-12-01:
I tried the v24 and the same problem happens. I can confirm that i did never update through Raspbian as you said!
BUT, if i log into Ubuntu Wayland, i can see the desktop! And then, if I login again in the normal Ubuntu everything is ok!

deltasierra-ufc, 2019-12-02:
tarsij,

   I did have this issue as well!  😲 I updated that text file as you mentioned before the close of your topic, rebooted, and the issue was corrected for me. 😃

   Thanks much!  Great work in finding that out.

   V/R,
Derek Swanson
IT Admin
Ursa Farmers Cooperative
Office: 217-964-2116
Mobile: 217-592-6468
dswanson@ursacoop.com
https://www.ursacoop.com
[1518816668039_PastedImage]

TheRemote, 2019-12-02:
I'm wondering the same.  I'm going to add in tarsij's fix to V25 but could you try it as well giulioturrisi?

TheRemote, 2019-12-04:
This has been fixed in v25 -- it implements the fix that tarsij shared which has fixed the issue that was affecting most people
If you are still having trouble in v25 please open another issue so we can keep track of them and not let anything slip through the cracks!

closed 2019-12-04 by TheRemote